qmk/qmk_firmware #7350: [Bug] Planck won't work on Linux

jbaum98, 2019-11-13:
Flashing the keyboard with any commit past cec391df7 breaks my planck rev6.

## Describe the Bug

Flashing my planck rev6 with the latest firmware causes it to not work on my laptop running NixOS. It does work on a friend's Windows machine. I've pinpointed the issue to cec391df7; removing this commit fixes the issue.

## System Information

 - Keyboard:  Placnk
   - Revision (if applicable):  6
 - Operating system: NixOS
 - AVR GCC version: 8.3.0

 - ARM GCC version:  6.3.1 20170620

 - QMK Firmware version: 0.7.66

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Myridium, 2019-11-14:
Good job narrowing it down to a specific commit. Are you able to narrow it down further? The only things that changed were preonic keyboard settings, and planck settings in the `config.h` and `rules.mk`. Try starting with all of the changes, and revert them line by line until it works. There are only 13 lines or so.

tschaerni, 2019-11-14:
looks like a duplicate of #7315.
Cheers~

zvecr, 2019-11-14:
Should be fixed by #7330, can you try updating please.

jbaum98, 2019-11-14:
Yup, all good now. Thanks for the quick fix!

closed 2019-11-14 by jbaum98

zvecr, 2019-11-14:
Thanks for testing!