qmk/qmk_firmware #7370: Consider adding Keychron K* keyboards

mdreizin, 2019-11-15:
I am a user of [Keychron K2](https://www.keychron.com/products/keychron-k2-mechanical-keyboard) keyboard I only know that it uses `ARM M0 MCU`. I am not an expert and it would be nice to know if it is possible to create QMK-based firmware for this keyboard or not?

yanfali, 2019-11-15:
@mdreizin someone will need to do some reverse engineering and figure out the vendor and bootloader.

mdreizin, 2019-11-15:
@yanfali Do you know what questions should I ask Keychron (about vendor/hardware/etc)?

I will try to send them directly (or their user group) and they would happy to answer/provide bootloader and other stuff.

mdreizin, 2019-11-16:
@yanfali I am looking forward to getting your reply 😀

timescam, 2019-11-17:
[Communication protocol in Chinese](https://drive.google.com/file/d/1pmjYJr2AP08ed-RSi8YZvZzSyYpqQ6Ni/view?fbclid=IwAR25cvJQHKl_7FyNcwNdvDl0VacN3Fy6aaLNoHJpCWMcO79FPvhuLko3x9w) for keychron keyboards, released on the fb group.
Hope this helps.

jkramarz, 2019-11-17:
@yanfali, according to flashing tool in firmware upgrade package, it is using Cortex M0 Sonix SN32F260. On the other side, chips inside (at last in white-LED version) are HFD48KP500 and CYW20730.

Flashing tool and firmware files can be found at https://drive.google.com/file/d/1F49mMZpGny9VWB-MmuwtZYJL-Soxhzc4/view?usp=sharing .

yanfali, 2019-11-18:
So the issue will be this isn't supported by any of the HAL layers we use. Currently ChiBIOS and ATSAM for ARM platforms. Therefore this is going to be a lot of work to add support for. You're going to have to find someone with the hardware willing to do the work of porting and supporting this chip. That's about all the assistance I can offer at this time.

amadeusrotundo, 2019-11-18:
> You're going to have to find someone with the hardware willing to do the work of porting and supporting this chip. 

Not knowing anything about ARM programming and the like, would you say it would probably take less effort to desolder the current ARM chip and put in a compatible one vs the reverse engineering of the current one?

noroadsleft, 2019-11-18:
> So the issue will be this isn't supported by any of the HAL layers we use. Currently ChiBIOS and ATSAM for ARM platforms. Therefore this is going to be a lot of work to add support for. You're going [...]

Actually, some of the CannonKeys boards run Cortex M0 chips currently, so I'm fairly sure ChibiOS could run this (supporting the chip being the barrier to entry, rather than the architecture).

zvecr, 2019-11-18:
When it comes to ChibiOS, its less about the M0 part and more if there is a hardware port for it. A quick search for sonix support doesn't show much hope, and I would guess ending up something like qmk/ChibiOS-Contrib/pull/10.

smp4488, 2019-12-14:
Not sure if this is off topic, but I’ve been looking into porting ChibiOS to the SN32F248B and it is possible that some of the work will carry over. I’m able to upload firmware and get into recovery mode. I would like to enable SWD on the current firmware to allow me to step through the code. Here is my progress so far. 

https://github.com/smp4488/dk63

xyzz, 2019-12-14:
Did you ever get the st-link (clone?) working with it? I see there are configs for rpi and stlink, but I guess only the rpi one ended up working since that's what all the pictures show?

smp4488, 2019-12-14:
@xyzz I did, i need to update the readme. I couldn't get the windows st-link to connect, but it did get it working with openocd.

xyzz, 2019-12-14:
Would appreciate it, I have a keyboard with a similar chip and I couldn't get it working with [blackmagic](https://github.com/blacksphere/blackmagic), haven't tried openocd though.

smp4488, 2019-12-14:
Give OpenOCD a shot with the st-link config in my repo. Should work with Sonix and SMT32 MCUs

xyzz, 2019-12-15:
Thanks, I got it working with both blackmagic and openocd now. Are you able to read memory/flash from gdb?

smp4488, 2019-12-15:
I can read memory past 0xFFF0000 but nothing below that while in debug recovery mode. I have the firmware and I’m trying to find the bits to enable SWD so I can read everything else.

xyzz, 2019-12-15:
I think this is a weird code protection function they use. Check the value of CS at 0x1FFF2000, it's probably going to be 0x5A5A (it was for me). Also, are you on discord? Can you add me, xyz#9668, so that we can iterate on it faster without having to spam the issue tracker.

ByteWoofer, 2020-01-29:
New [Keychron K4](https://www.keychron.com/products/keychron-k4-mechanical-keyboard)  RGB user here. Is there any assistance that I could provide to this issue? I am semi competent with hardware hacking and willing to take an [Attify badge](https://blog.attify.com/hack-iot-device/)  to the onboard flash and assist in analysis to the best of my abilities.

smp4488, 2020-01-29:
Hello @ByteWoofer, thank you for wanting to offer your assistance. With the help of @xyzz we were able to get chibios running with limited USB. I’m working on flushing out GPIO right now. You probably want to identify your chip and start setting up board and config files for your keyboard. 

Here are the repos we are currently working out of. 

QMK - https://github.com/smp4488/qmk-ChibiOS/tree/dk63?files=1

ChibiOS - https://github.com/smp4488/qmk_firmware/tree/chibios-upgrade?files=1

StrayChild01, 2020-01-29:
> Hello @ByteWoofer, thank you for wanting to offer your assistance. With the help of @xyzz we were able to get chibios running with limited USB. I’m working on flushing out GPIO right now. You [...]

Hi Guys, another K2 user here. I stumbled on this issue and I remember that in facebook one of the core team members of keychron was actually asking what was needed to get the keyboard working with QMK. Maybe we can ask him to see if he can provide more information about the chips and the instruction set🤔

kusayuzayushko, 2020-01-31:
Hello. [Redragon K585 DITI](https://www.redragonzone.com/products/redragon-k585-keyboard) user here. it has a VS11K16A on board. 
Also looking for a way to reflash it with QMK or ChibiOS. Any info i can provide?

fightforlife, 2020-02-04:
For someone googling this.
The Womier K66 is also using this chip. And wants to be ported to qmk :)

More info: https://www.reddit.com/r/olkb/comments/eqfa68/womier_k66_qmk_porting_possible_vs11k16avs12l03a/

xyzz, 2020-02-04:
Womier K66 is in fact one of the keyboards being working on in this branch at https://github.com/smp4488/qmk_firmware/tree/chibios-upgrade/keyboards/womier/k66

ByteWoofer, 2020-02-04:
> Hello @ByteWoofer, thank you for wanting to offer your assistance. With the help of @xyzz we were able to get chibios running with limited USB. I’m working on flushing out GPIO right now. You [...]

Just finally got a moment away from work to open up the keyboard. The chips inside are as follows:
[image]
[image]
[image]
[image]

smp4488, 2020-02-05:
@ByteWoofer so I started looking for the HFD2201KBA and came across this interesting tweet https://twitter.com/pmg/status/1119008876240596992?lang=en. Your chip looks to be a rebranded STM32 chip. Mine is rebranded Sonix chip. I would start trying to identify some pins on your MCU and compare to some STM32 variant data sheets to find a match. 

You also appear to have a JTAG port on the bottom of the #2 picture. https://blog.senr.io/blog/jtag-explained

fightforlife, 2020-02-09:
> Womier K66 is in fact one of the keyboards being working on in this branch at https://github.com/smp4488/qmk_firmware/tree/chibios-upgrade/keyboards/womier/k66

How are you currently talking to the Womier K66? Did you find a way to use the USB Bootloader mentioned in the spec sheet.

xyzz, 2020-02-09:
For flashing we're using the built-in bootloader which isn't ideal but works well enough. Getting into the bootloader is possible by shorting the boot pin to ground or sending a special hid command to the stock firmware.

fightforlife, 2020-02-10:
Is it possible to include this information into the K66 QMK Readme file. 
I think this would help to get people started tinkering with this board.
Thanks!

P.S.: Is it possible to save/dump the original firmware and reflash it later? Or is it protected?

xyzz, 2020-02-10:
Currently it's somewhat dangerous to flash and a pain to recover if something goes wrong. The tooling is also only tested on Linux so far. That said if you're ok with the limitations (eg lack of rgb in qmk), ping me in mechkeys discord and I can guide you through the process.

As for the original firmware it's possible to flash it back yes.

xyzz, 2020-02-24:
I've now posted a (tested) flashing guide at https://gist.github.com/xyzz/471d5856376cdce945e686785c75f7cf

justMaku, 2020-02-27:
I have just became a quite happy owner of the Keychron K2 but obviously I'd like to use QMK with it. I have looked at the macOS upgrade utility and I have successfully decrypted the enclosed firmware.

I'm unsure if this applies also to Linux and Windows updaters, but on macOS, the update data is contained within `UserInfo.plist` file and encrypted using AES 128 in ECB mode. The key is also contained within the plist file (under the `token`) field and right now uses the secure random value of `123456`.

After decrypting, the resulting file is then `lzfse` compressed (Apple's compression mechanism) and after decompressing you're left with standard intel hex file.

StrayChild01, 2020-02-28:
> I have just became a quite happy owner of the Keychron K2 but obviously I'd like to use QMK with it. I have looked at the macOS upgrade utility and I have successfully decrypted the enclosed [...]

After seeing this, I wondered if this happens as well on windows. And it actually is easier to get the firmware!

I opened the "K2-RGB-V1.6_En for Windows.exe" file using 7zip and voilá:

The contents are:
/FWFiles
/Image
UISettings.ini
UISettings_user_guide.txt

I checked opened the .ini and .txt files and I can see the content. I also was able to get the .hex file.

meanderer-tech, 2020-04-27:
> [Communication protocol in Chinese](https://drive.google.com/file/d/1pmjYJr2AP08ed-RSi8YZvZzSyYpqQ6Ni/view?fbclid=IwAR25cvJQHKl_7FyNcwNdvDl0VacN3Fy6aaLNoHJpCWMcO79FPvhuLko3x9w) for keychron keyboards, released on the fb group.
> Hope this helps.

Is there a need to translate this document?

tzarc, 2020-04-27:
Only if the intent is for people to maintain the SN32 hardware port in the official ChibiOS-Contrib.

pgillan145, 2020-06-08:
I've been working on getting this installed on my Womier K66.  So far I've gotten the chibios-upgrade branch from @smp4488 to compile (with a minor change to the ram size settings in SN32F260.ld), and basic keymapping works, but not mouse or media controls.

I also got @toast's womier-k66-rbg-support branch to compile, but installing it just locks up the keyboard (though with rainbow colored lights).

flashquark, 2020-06-18:
> > Hello @ByteWoofer, thank you for wanting to offer your assistance. With the help of @xyzz we were able to get chibios running with limited USB. I’m working on flushing out GPIO right now. You probably want to identify your chip and start setting up board and config files for your keyboard.
> > Here are the repos we are currently working out of.
> > QMK - https://github.com/smp4488/qmk-ChibiOS/tree/dk63?files=1
> > ChibiOS - https://github.com/smp4488/qmk_firmware/tree/chibios-upgrade?files=1
> 
> Just finally got a moment away from work to open up the keyboard. The chips inside are as follows:
> **Chip 1**
> [image]
> **Chip 2**
> [image]
> **Chip 3**
> [image]
> **Chip 4**
> [image]
> 
> **Whole board**
> [image]

In case you were still wondering, the HFD2201KBA is also a rebranded SN32F248B. 
I figured this out by using Resource Hacker to check inside a keyboard firmware flashing utility for the Velocifire M2 60% keyboard, which has the HFD2201KBA chip. Inside, it references SN32F240B_32.hex for the firmware hex file which means that it is definitely in the SN32F240B series of chips. 

[image]

Looking up the chip series on Sonix's website: http://www.sonix.com.tw/list-en-4336

shows that there are 4 chips in this series, but only the SN32F248B comes in the LQFP64 package, which is what the HFD2201KBA is, so it is almost certainly this chip.

As for the bluetooth module, it is a rebranded Broadcom BCM20730 as can be found here: https://www.cypress.com/file/298211/download

1Conan, 2020-07-08:
Can confirm my findings is inline with @flashquark
Pinouts on the keyboard I have is similar with the SN32F240 series. havent extensively tested the SWD pads on mine though.

ref: https://twitter.com/pmg/status/1220131984678125568

tiena2cva, 2020-07-15:
Is this issue still open? Have any working in this? I want to join.

tzarc, 2020-07-15:
It can stay open as long as people are willing to follow:
https://docs.qmk.fm/#/platformdev_selecting_arm_mcu?id=add-new-mcu-family

If there's nobody willing to do so, then it can get closed.

LazaroFilm, 2020-07-18:
+1 for interest, I have a K2 and the lack of layout customization is really annoying. Would love for QMK to work with it.

Nimayer, 2020-07-19:
+1 for interest.
I have a K2 and with the stock firmware the Fn keys don't work on Linux :(
I have some embedded experience and would be available for contributing.

garrett, 2020-07-19:
@Nimayer for using a K2 on Linux, check out https://github.com/Kurgol/keychron/blob/master/k2.md

The Fn keys not working on Linux is related to running `echo 0 | tee /sys/module/hid_apple/parameters/fnmode` (as root). There's [a udev line](https://github.com/Kurgol/keychron/blob/master/k2.md#udev) on that page to have it auto-set the keyboard in the correct mode, running this command behind the scenes when you connect the keyboard. Alternatively, there's a modprobe config setting to change the fnmode too.

As for the stock firmware, it can only be updated with Windows and macOS at the moment. There's a rumor that Keychron is considering working with [LVFS](https://fwupd.org/) to get fwupd support... but I haven't found any actual evidence of it. (And have meanwhile updated my K2 using Windows.)

This all said, I'd love to see qmk firmware support for the Keychron K2. 

FWIW: There's [a similar effort for the Anne Pro 2](https://openannepro.github.io/), another currently popular keyboard... and they have it all working already. Being able to have qmk on a pre-made keyboard (opposed to having to know how to solder or find a hotswap PCB somewhere and assemble a keyboard one's self) is a huge deal.

fronbasal, 2020-07-24:
+1 for interest
I have a K4 and I'd love to use QMK on it!

Eltherington, 2020-07-28:
Another +1 for interest. I have a K6. 
If there is anything I can do to help, anything needed, let me know.

tiena2cva, 2020-07-28:
+1 for interest

luukvankooten, 2020-07-28:
[Game keyboard communication protocol.pdf](https://github.com/qmk/qmk_firmware/files/4989172/Game.keyboard.communication.protocol.pdf)
Thanks to google translate. The communication protocol in English. I hope that it helps. I also have interest +1.

frankk74, 2020-08-11:
+1 for k2 interest as well... thanks to everyone so far :-) have k2 I got 1 month ago ... wish I knew then it wasn't "open"... closed source :-(

vladimir-aubrecht, 2020-08-16:
+1 for K8 support! 😊

danhnguyenncd, 2020-08-20:
+1 for Keycool KC84 rgb ( ver blue 5.0)

sakura-he, 2020-08-20:
+1 for Keycool KC84 ( ver blue 5.0)

Fuzen-py, 2020-08-25:
+1 for K1 support

joshfriend, 2020-08-25:
Please everyone stop adding +1 comments. Instead, click the :+1: reaction emoji on the toplevel post:

[image]

Adding "+1" comments only adds noise to the discussion of the folks trying to move this feature forward

digdog, 2020-09-09:
There're many data sheet, sw and dev tools available for download from Sonix's Trad. Chinese pages: 
http://www.sonix.com.tw/article-tw-4315-30347 (SN32F248B)
http://www.sonix.com.tw/article-tw-4315-30346 (SN32F247B)
http://www.sonix.com.tw/article-tw-4315-30345 (SN32F246B)
http://www.sonix.com.tw/article-tw-4315-30348 (SN32F2451B)

Raqbit, 2020-09-09:
@digdog I must say, NuPhy's site looks... familiar :smile: 

Perhaps some of the people working on the NuType are/were involved with both products? (Or they may have taken inspiration from Keychron)

[image]

----

The Windows updater for my Keychron K4 RGB w/ Optical switches contains firmware (Intel hex) for the "SN32F260":

```ini
[SpecialPID0]
VID=0x05AC
PID=0x024F # It's pretending to be an Apple USB keyboard
ChipName=SN32F26x
FWName=SN32F260.hex
FWType=HEX
CodeSecurity=1
```

So it looks like the Keychron K4 RGB-Optical is using the same chip as the Wormier K66 RGB, as far as I can tell.

MingyaoLiu, 2020-09-17:
The Drevo Seer Pro 60% is using the SN32F240B_32.hex firmware.
File inside updater file  states:

```[UISettings]
ISPTitle=DREVO Seer PRO Firmware Update_FW V1.10
ISPProgramText=ISP Program
ISPWarningMSG=
ISPWarningMSG2=
ISPMSGFontColorR=255
ISPMSGFontColorG=255
ISPMSGFontColorB=0
ISPMSGFontColorSize=25
StartButtonString=Start
ISPStartMSG=Start update...
ISPSuccessMSG=Update successfully
ISPFailMSG=Update fail
ProgramOneTime=0
ACPowerCheck=0
ACPowerCheckWarningMsg=Please connect AC-adapter
CheckDeviceCmd=AA42895AFF7162CC
CheckBootLoaderID=1
DeviceChangeCheck=0
CurVersion=0x0109
ISPFWUpdatedMsg=Firmware is already updated
ShowMessageBox=0

[SpecialPID]
Count=3							
					
[SpecialPID0]						
VID=0x0C45						
PID=0x8006						
ChipName=SN32F24xB					
FWName=SN32F240B_32.hex			
FWType=HEX						
CodeSecurity=1						

[SpecialPID1]
VID=0x05AC						
PID=0x024F						
ChipName=SN32F24xB					
FWName=SN32F240B_32.hex				
FWType=HEX						
CodeSecurity=1	

[SpecialPID2]
VID=0x05AC						
PID=0x0256						
ChipName=SN32F24xB					
FWName=SN32F240B_32.hex				
FWType=HEX						
CodeSecurity=1			
```

I really want to remap the Fn key, but have no luck using the Drevo Power Console software.

ervinpm, 2020-10-21:
+1 would like to see this working for the velocifire m2. I wonder if asking the manufacturer for source codes would help?

stdvar, 2020-10-21:
> +1 would like to see this working for the velocifire m2. I wonder if asking the manufacturer for source codes would help?

No hope. Asked. They don’t have plans for making boards configurable. Nor they will release sources.

SmileYzn, 2020-10-24:
MotoSpeed CK62 is very similiar to TKL61WS but TKL61WS as a lot of better layer than CK62.

Here is the link for TKL61WS  : https://www.velocifiretech.com/product/m2-wireless-mechanical-keyboard/

teknogeek, 2020-10-24:
What's very surprising to me is that I just launched Drevo Power Console with my Keychron K6 plugged in, and it recognized it and even let me change RGB settings (?!):

[image]

However, it seemed to mess up the RGB profiles and caused the "switch profile" button to forget all the RBG profiles but the one set in the software:

[image]

I wouldn't really recommend trying this, since I had to reflash [the OEM firmware](https://www.keychron.com/pages/firmware-for-keychron-k6) to get RGB working properly again.

teknogeek, 2020-11-15:
Also, the same behavior above is seen when using the [Ninja71 keyboard software](https://www.monstargears.com/75/?q=YToxOntzOjEyOiJrZXl3b3JkX3R5cGUiO3M6MzoiYWxsIjt9&bmode=view&idx=3106747&t=board) as well

[91 more replies]

closed 2022-08-29 by tzarc