qmk/qmk_firmware #7408: [Bug] Broken link in docs/other_eclipse.md

mtei, 2019-11-19:
## Describe the Bug

In the `docs/other_eclipse.md`:

```markdown
# Prerequisites
## Build Environment
Before starting, you must have followed the [Getting Started](README.md#getting-started) section corresponding to your system. In particular, you must have been able to build the firmware with [the `make` command](../#the-make-command).
```

There is two link seemd broken:
 * `README.md#getting-started`
 * `../#the-make-command`

## System Information

 - QMK Firmware version: 0.7.79

mtei, 2020-01-07:
Does anyone know what the right link is?

fauxpark, 2020-01-08:
I think this sentence should just be reworded to point at the newbs guide instead.

noroadsleft, 2020-01-10:
I agree with @fauxpark – just rework it to point at https://docs.qmk.fm/#/newbs_getting_started.

mtei, 2020-03-14:
Correction completed. 👍

closed 2020-03-14 by mtei