qmk/qmk_firmware #7411: Storage : QMK global storage

zekth, 2019-11-19:
After discussing with @skullydazed and @yanfali about storing the user data for the configurator one idea came in. A global repository to handle the storage of any qmk project. The qmk project would be the skeleton of the storage and users will fork it to use their data.

So for example the structure would be : 
```
|-keymaps
  |-foo
    |-bar.c
|-AnyOtherProject
  |-datatostore
    |-fileToStore.json
```

This repository will be forked by user and can be added as submodule in the QMK_firmware project. This process of forking / updating the fork can be also handled by the `qmk_api`

So an example of usage is having the list of your keymaps available in the `QMK_configurator` while you're connected to it using Github.

skullydazed, 2019-11-19:
I would like to highlight being able to add this repo as a submodule- this seems like a powerful pattern and I wish I had thought of it myself.

skullydazed, 2019-11-19:
Some thoughts-

Keymaps will have to be limited to JSON, so that configurator can easily parse them. Parsing C files has proven to be error-prone so far.

It's not clear what `AnyOtherProject/datatostore` is. How are these different from keymaps?

I think it would be better for the github forking/storage to be handled in the browser. This way we don't have to handle user credentials server side and there is less operational overhead.

zekth, 2019-11-19:
> It's not clear what `AnyOtherProject/datatostore` is. How are these different from keymaps?

At the moment we need it for keymaps, but after we may need this storage for another purpose.

> I think it would be better for the github forking/storage to be handled in the browser. This way we don't have to handle user credentials server side and there is less operational overhead.

This was also my intention

skullydazed, 2019-11-19:
We've been kicking around the idea of making userspace fully self-contained, as described in #7414. If we follow the same tree structure this repo could simply be cloned or submodule'd under `qmk_firmware/users`.

closed 2019-11-19 by zekth