qmk/qmk_firmware #7415: [Bug] MT not working as expected

stevexyz, 2019-11-19:
The “jj28” (jj40/stevexyz) is one of the keymaps that put the framework to the limits, and I’ve found a strange behavior that in my opinion can be considered to be a bug (for sure is not as the keyboard is expected to react and so a solution should be found).

To enter in detail,  this is the setup:
1. Space key configured as MT(MOD_RSFT, KC_SPC)
2. “A” key (can be any other key) configured simply as KC_A

With this configuration if you press and release A key, and later on you press space and before releasing it (and within a short period) you press again A and release both, the result expected is “aA”.

And this is what is happening if you let time pass between the first “a” and the second “A”.

The problem appears if “shift  a” is being executed (quickly) just after the first “a” is being released; the result is then an unexpected “a a“ (meaning the space worked as key instead as shift)…

bcat, 2019-11-20:
I've never understood _why_, but this is expected default behavior. You can set `TAPPING_FORCE_HOLD` to change this (but you lose the ability to repeat the tap code, and I think this also breaks tap dance somehow).

stevexyz, 2019-11-22:
@bcat yes, you are completely right!!

I don't need tapping toggles, but unfortunately for how it is designed the keyboard now I've lost the possibility to autorepeat keys... But thinking of it seems to me there is no option and this is a limitation that I should live with.

Thank you!

closed 2019-11-22 by stevexyz