qmk/qmk_firmware #7436: Rotary encoder switch pin

skeltr, 2019-11-21:
I want to connect the switch pin of my rotary encoder to do a GUI on an OLED like this : https://www.youtube.com/watch?v=S8yhJG7dxO8&list=PL_HwU1Q1f7IYvV4kUGnG04anpTefFd6yX&index=5
But I don't know if I have to connect it to the matrix, define a direct_pin or a custom matrix.
Can you help me please ?

noroadsleft, 2019-11-21:
@XScorpion2 wrote that code (that's actually his YouTube video you've linked).

I believe the code is split between:

- [XScorpion2's userspace](https://github.com/qmk/qmk_firmware/tree/f0790a722d5a586e3621487b554fd217839c24c0/users/xulkal) (the `custom_encoder` and `custom_oled` files, mostly)
- [RGBKB Sol codebase](https://github.com/qmk/qmk_firmware/tree/f0790a722d5a586e3621487b554fd217839c24c0/keyboards/rgbkb/sol)

drashna, 2019-11-21:
^ . This.  

Also, the two features that are being used here are: 
* https://docs.qmk.fm/#/feature_encoders
* https://docs.qmk.fm/#/feature_oled_driver

Besides that, you'd have to write all the code to do the interaction.

closed 2019-12-01 by skeltr