qmk/qmk_firmware #7479: ChibiOS (Proton C) Adafruit BLE Support

joshuarubin, 2019-11-25:
I'd like to start working on adding support for the adafruit ble to the proton c (chibios generally). I was just hoping to get some pointers for what that should look like.

* I've started playing around with the HAL SPI driver
* I know that there is no adafruit library for arm, so I'd have to replicate that. Will use the [SDEP Transport Protocol](https://learn.adafruit.com/introducing-the-adafruit-bluefruit-spi-breakout/sdep-spi-data-transport)
* According to #912, it would probably be nice _"to have a 'proper' QMK SPI interface that does the right thing"_. As I'm new to this project I'd probably need some guidance on what that means. Otherwise, I would probably just make this avr specific somewhere in `tmk_core/protocol/chibios`.
* `config_common.h` errors out if `BLUETOOTH_ENABLE` is defined (I guess it becomes defined when I set `BLUETOOTH = AdafruitBLE`) on the proton c because it can only configure the USART for a few chips (e.g. `__AVR_ATmega32U4__`). We don't need USART here, but what definition should I use to identify the proton c in order not to error out here?
* Where can I find an electrical schematic of the proton c?
* `stm32/mcuconf.h` will need one of `STM32_SPI_USE_SPI1/2/3` enabled, but I don't know which one.

I'm sure I'll have more questions. Thanks for the help!

yulei, 2019-11-28:
I have ported the adafruit ble to chibios before,(https://github.com/yulei/qmk_firmware/tree/wm690/tmk_core/protocol/chibios), you can check the spi.h.c, adafruit_ble.h.cpp. I'm using a stm32f4(proton c used stm32f3), and you can try to port them to using stm32f3.

joshuarubin, 2019-11-28:
That's great, thank you!

One thing I'm stuck on is, since I have a battery (it will stay on when I remove the USB cable), I would like to be able to unplug the usb, have it switch to bluetooth (this part I can get, I look for the `USB_ACTIVE` state, ironically, this seems wrong to me...), and when I plug in the usb again, it would switch back. However, once I unplug the usb, I can't get it to reconnect with the computer when plugging it back in. I have to shut down the board manually. Is it possible to have the USB reconfigured when it is reattached?

yulei, 2019-11-28:
you need to detect the presence of the VBUS, that means a dedicate IO pin will be used, once you found the VBUS power was lost. Stop and release the usb driver, and re-initialize the usb driver while VBUS was power on.  Or just reset the MCU to re-configure the USB.

joshuarubin, 2019-11-28:
How do I reset the MCU without going back into the bootloader?

Similarly, how is the usb driver released and reinstalled?

I think I just need to be pointed in the right direction. Should I be looking at chibios docs? The MCU docs?

smp4488, 2020-03-23:
I am also interested in getting SPI working with chibios for a different Bluetooth module. I’m looking for guidance on how this should be structured. I don’t have a chibios SPI driver yet but I was thinking about setting this up with out it.

fauxpark, 2020-03-23:
@smp4488 #8299 will provide the interface for a potential ARM SPI driver - it's similar to the existing i2c_master stuff.

tzarc, 2020-04-08:
I've merged the AVR spi_master implementation for #8299.
We'll need to sort out an equivalent ChibiOS implementation.

tzarc, 2020-04-09:
Have raised a draft PR, #8739, which lays the groundwork for the SPI protocol on ChibiOS/ARM.
The Adafruit BLE will require a bunch of changes to turn it into a generic driver after that PR gets merged.

closed 2020-08-07 by stale[bot]