qmk/qmk_firmware #7487: [Bug] Encoder support causing spamming on iris v3

yanfali, 2019-11-26:
@XScorpion2 patch 0f0c73f14a8ffb83a79b51582c0e7465c2411749 on the 15th is causing spamming on the right side of an Iris v3 with default keymap on Page UP and Page Down keycodes. This is an unpopulated PCB. The PCB has no encoder on the right side, only the left side. Seems like the code is making some assumptions about the presence of an encoder.

## Describe the Bug
Before 0f0c73f14a8ffb83a79b51582c0e7465c2411749 everything is fine, after that patch the spamming occurs.

To reproduce:

 - flash commit before - no spam
 - flash commit 0f0c73f14a8ffb83a79b51582c0e7465c2411749 - spam

## System Information

 - Keyboard:  Iris
   - Revision (if applicable):  rev 3
 - Operating system: MacOS Mojave
 - AVR GCC version: 8.3.0

 - ARM GCC version: N/A

 - QMK Firmware version: N/A

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

yanfali, 2019-11-26:
Confirmed independently by @nooges

yanfali, 2019-11-26:
@XScorpion2 since you wrote this patch, do you want to take a stab at diagnosing what's going on? Thanks

XScorpion2, 2019-11-26:
When you say spamming, is it spamming in one direction? or alternating? or what?

nooges, 2019-11-26:
I'm getting just PGDN spammed on my board.

yanfali, 2019-11-26:
yep, if you put it in into https://config.qmk.fm/#test you will see the PGDN key being repeatedly sent. This is just right right half when plugged in. Also happens when you plug in both halves and have trrs between them.

nooges, 2019-11-27:
Possibly related: https://www.reddit.com/r/olkb/comments/e0enua/flashing_iris_rev_4

nooges, 2019-11-27:
Okay, getting the same PGDN spamming on Iris Rev. 4 right PCB as well, no spamming with left half.

yanfali, 2019-11-27:
Should we revert this patch?

XScorpion2, 2019-11-27:
Probably should revert it. Another users Sol r1 also is showing this issue. I cannot repro this locally so it's taking some time to fix.

yanfali, 2019-11-27:
interesting, while reverting this, I got into a conflict

**<<<<<<< HEAD
static uint8_t encoder_state[NUMBER_OF_ENCODERS]  = {0};
static int8_t  encoder_pulses[NUMBER_OF_ENCODERS] = {0};
=======
static uint8_t encoder_state[NUMBER_OF_ENCODERS] = {0};
>>>>>>> parent of 0f0c73f14... Updated slave encoder sync to reduce dropped pulses (#7325)**

changing the data type might explain what happened here

yanfali, 2019-11-27:
I switched it to uint but no change

XScorpion2, 2019-11-27:
Don't switch thos types. Those are required for the delta calculation to work correctly.

yanfali, 2019-11-27:
> Don't switch those types. Those are required for the delta calculation to work correctly.

Can you send a PR to revert this temporarily while you fix it? I tried to do it automatically but now it's too deeply embedded

XScorpion2, 2019-11-27:
Will do

XScorpion2, 2019-11-27:
@yanfali https://github.com/qmk/qmk_firmware/pull/7498

yanfali, 2019-11-28:
https://github.com/qmk/qmk_firmware/issues/7055 is being re-opened because we had to revert that fix. Closing this for now.

closed 2019-11-28 by yanfali