qmk/qmk_firmware #7511: BL_ON turns the LEDs off and BL_OFF turns the LEDs on

HMarxen, 2019-11-29:
Hi, I have made a board with a NPN transistor between the LEDs and ground. For some reason BL_ON and BL_OFF work in reverse. How can I reverse the behaviour of the backlight pin so that the pin goes high when it would normally go low and the other way around? What puzzles me is that this seems to be the standard method to do backlighting. So, others must have the same problem.

HMarxen, 2019-11-29:
It seems that at full brightness QMK PWMs the backlight pin still a little bit. So, another bad side-effect is that my LEDs never turn completely off. They are always glowing very dimly.

noroadsleft, 2019-11-29:
@fauxpark, any thoughts? Is this a BACKLIGHT_ON_STATE issue?

HMarxen, 2019-11-29:
Yes, Thank You. I don't know why I overlooked that when I was brooding over the documentation.
But what is really strange is that my NPN transistor behaves in reverse.
I had to use this strange one because it is squeezed in a tight spot:
https://www.mouser.de/datasheet/2/115/ZTX1049A-23889.pdf
looks standard NPN to me.

fauxpark, 2019-11-29:
I'm going to guess you're not using a hardware PWM pin (those are listed in the docs). Currently, when the backlight transistor is connected to a hardware PWM pin, the behaviour is correct *only* if it's NPN, as it's hardcoded. When connected to any other pin, `BACKLIGHT_ON_STATE` comes into effect. Unfortunately the default value is 0, meaning QMK will output a logical *low* to the transistor when it wants to turn the backlight on, and vice versa (for a PNP transistor).

I have a PR that consolidates this behaviour, however it has to go through the breaking changes process for a couple of reasons: #6749

Basically, for now you'll need to `#define BACKLIGHT_ON_STATE 1` in your config.h. Eventually this will become the default.

HMarxen, 2019-11-30:
Ah that is good to know. Yes, it is software PWM. In the next revision it will be hardware PWM. I didn't know at the time that timer 0 was used internally by QMK.

closed 2019-11-30 by HMarxen