qmk/qmk_firmware #7524: can't compile Chibios I2CV1 driver under stm32f4 through QMK configurator

yulei, 2019-12-02:
While I compile the keyboard (matrix/noah), the following error occurs:
[image]

But the latest qmk  compiled successfully under my local machine, and I can't find " qEvt(0x0000);" on the lib/chibios/os/hal/ports/STM32/LLD/I2Cv1/hal_i2c_lld.c. I also checked chibios 17, 18 and 19, but they all did not contains such codes.

The cannonkeys/ortho48 also use the I2CV1, and it compiled successfully through the qmk configurator.

yanfali, 2020-02-12:
https://github.com/qmk/qmk_configurator/issues/655

yanfali, 2020-02-12:
Something's weird. This is either a regression or it's not fixed.

zvecr, 2020-02-22:
[image]

Seems fine now so im going to close this, feel free to reopen if you still see the issue.

PR #7526 (2019-12-13): fixed I2C driver support for stm32f4

stm32f4 uses I2Cv1 and GPIOv2 drivers, so the current definitions on i2c_master.h should be modified.

## Description

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #7524 

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I hav [... cut]
closed 2020-02-22 by zvecr