qmk/qmk_firmware #7546: Is it possible to have a second hardware PWM pin with its own controls?

HMarxen, 2019-12-04:
Hi,
I would like to add a TLC6C598PWR IC to my design which is a shift register with a seven-segment display driver. The IC has an enable pin which allows to PWM the segment display. So, it would be fantastic if I could set an independent brightness for the display. On the 32u4 I would hook the key backlight to the timer 1 and the segment display to timer 3. 
If this is theoretically possible, this would also be a feature request. I don’t think I could contribute the code myself. :-/

HMarxen, 2019-12-07:
With a second hardware PWM you could also put a transistor to the common cathode of the num lock, scroll lock and caps lock LEDs and adjust their brightness independently. Or you could try other cool stuff like using an analogue 5 volt meter as a keystroke per minute gauge. The faster you type the more goes the physical needle into the red.

drashna, 2019-12-13:
Do you mean for the STM32 chips?

HMarxen, 2019-12-13:
No the Atmega32u4. Would this be easier to do on a STM32?

HMarxen, 2019-12-17:
Anyoune?

noroadsleft, 2020-03-17:
I'm neither a hardware guy nor a programmer, so take this with a grain of salt, but:

32U4 seems to have four PWM channels. On those chips, I believe Backlight uses one channel and Audio uses a second, which leaves two channels unused. Seems the limitation is writing the supporting programming.

sigprof, 2020-06-26:
More precisely, ATmega32U4 has 4 timers; however, Timer0 has only 8 bits and is used as the system timer, and Timer4 is a special “high speed timer” (it is 10 bit with a possibilty of using 11 bit resolution for PWM). The current backlight code is able to use only Timer1 and Timer3, which are 16 bit timers; however, each timer output can be connected to a limited set of pins. It could be possible to use a single timer for up to 3 PWM signals (which will have the same frequency, but different duty cycles), but only if the LEDs are connected to specific pins.

Pins that are usable for PWM are:
- `B7` — Timer0 channel A (not usable in QMK), Timer1 channel C
- `B6` — Timer1 channel B, Timer4 channel B
- `B5` — Timer1 channel A, Timer4 channel B (complementary)
- `C7` — Timer4 channel A
- `C6` — Timer3 channel A, Timer4 channel A (complementary)
- `D7` — Timer4 channel D
- `D6` — Timer4 channel D (complementary)
- `D0` — Timer0 channel B (technically it could be used as a 8-bit PWM while keeping Timer0 running at 1000 Hz using the compare register A, but currently this is not implemented)

With the current code only B7, B6, B5, C6 pins are supported; using B7+B6+B5 at the same time could be possible with just one timer.

closed 2022-08-28 by zvecr