qmk/qmk_firmware #7551: [Feature Request] Combo mod hold key

tomwang2011, 2019-12-05:
Currently our mod keys options are the normal ctrl/alt/gui/shift as well as meh/hyper. However if I want to have a button to hold on 2 keys (alt+gui for example) I don't know if there is a way for that.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I know there is a way to do `LALT(LGUI(key))` but my use case is: on my mac I need to hold alt+gui then press left/right/up/down to snap the window to the part of the screen, instead of having 4 keys for it, I am hoping there will be a way to dedicate a button to alt+gui then I can press the arrows easily.

ridingqwerty, 2019-12-05:
You can wrap around `KC_NO` using those mod wrappers, e.g.:

`LALT(LGUI(KC_NO))`

tomwang2011, 2019-12-06:
it works, thanks for the workaround!

closed 2019-12-06 by tomwang2011