qmk/qmk_firmware #7574: ARM I2C driver not implemented for KINETIS MCU's (i.e. Teensy 3.2/3.5/3.6/LC)

St0RM53, 2019-12-09:
The I2C driver used for OLED functionality does not support KINETIS ARM microcontrollers.

## Feature Request Type

- [ X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I have been waiting to implement this functionality on my finished keyboard i made about 3 months ago. This has been discussed on the discord server and while some process was made, the driver has no proper definitions to compile and work with the Chibios hal driver. 

Setting the following in:
**mcuconf.h** 
```
#define KINETIS_I2C_USE_I2C1                TRUE
#define KINETIS_I2C_I2C1_PRIORITY           4
#define USE_I2CV1
#define I2C_DRIVER I2CD1
```

**config.h**
```
#define I2C1_OPMODE 1
#define I2C1_CLOCK_SPEED 100000
#define I2C1_DUTY_CYCLE 1

#define I2C1_SCL_BANK GPIOC
#define I2C1_SDA_BANK GPIOC
#define I2C1_SCL 10
#define I2C1_SDA 11
```

will result in the following compilation output:

```
tnf@tnf-VirtualBox:~/Desktop/github/qmk_firmware$ make lilith101:default
QMK Firmware 0.7.97
Making lilith101 with keymap default

arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  34376	      0	  34376	   8648	.build/lilith101_default.hex

Compiling: drivers/arm/i2c_master.c                                                                In file included from <command-line>:0:0:
./keyboards/lilith101/config.h:120:26: error: excess elements in struct initializer [-Werror]
 #define I2C1_CLOCK_SPEED 100000
                          ^
drivers/arm/i2c_master.c:38:5: note: in expansion of macro 'I2C1_CLOCK_SPEED'
     I2C1_CLOCK_SPEED,
     ^~~~~~~~~~~~~~~~
./keyboards/lilith101/config.h:120:26: note: (near initialization for 'i2cconfig')
 #define I2C1_CLOCK_SPEED 100000
                          ^
drivers/arm/i2c_master.c:38:5: note: in expansion of macro 'I2C1_CLOCK_SPEED'
     I2C1_CLOCK_SPEED,
     ^~~~~~~~~~~~~~~~
./keyboards/lilith101/config.h:121:25: error: excess elements in struct initializer [-Werror]
 #define I2C1_DUTY_CYCLE 1
                         ^
drivers/arm/i2c_master.c:39:5: note: in expansion of macro 'I2C1_DUTY_CYCLE'
     I2C1_DUTY_CYCLE,
     ^~~~~~~~~~~~~~~
./keyboards/lilith101/config.h:121:25: note: (near initialization for 'i2cconfig')
 #define I2C1_DUTY_CYCLE 1
                         ^
drivers/arm/i2c_master.c:39:5: note: in expansion of macro 'I2C1_DUTY_CYCLE'
     I2C1_DUTY_CYCLE,
     ^~~~~~~~~~~~~~~
In file included from ./lib/chibios/os/hal/include/hal_pal.h:129:0,
                 from ./lib/chibios/os/hal/include/hal.h:124,
                 from drivers/arm/i2c_master.h:28,
                 from drivers/arm/i2c_master.c:28:
drivers/arm/i2c_master.c: In function 'i2c_init':
drivers/arm/i2c_master.c:66:44: error: 'PAL_MODE_STM32_ALTERNATE_OPENDRAIN' undeclared (first use in this function)
     palSetPadMode(I2C1_SCL_BANK, I2C1_SCL, PAL_MODE_STM32_ALTERNATE_OPENDRAIN);
                                            ^
./lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_pal_lld.h:394:36: note: in definition of macro 'pal_lld_setpadmode'
     _pal_lld_setpadmode(port, pad, mode)
                                    ^~~~
drivers/arm/i2c_master.c:66:5: note: in expansion of macro 'palSetPadMode'
     palSetPadMode(I2C1_SCL_BANK, I2C1_SCL, PAL_MODE_STM32_ALTERNATE_OPENDRAIN);
     ^~~~~~~~~~~~~
drivers/arm/i2c_master.c:66:44: note: each undeclared identifier is reported only once for each function it appears in
     palSetPadMode(I2C1_SCL_BANK, I2C1_SCL, PAL_MODE_STM32_ALTERNATE_OPENDRAIN);
                                            ^
./lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_pal_lld.h:394:36: note: in definition of macro 'pal_lld_setpadmode'
     _pal_lld_setpadmode(port, pad, mode)
                                    ^~~~
drivers/arm/i2c_master.c:66:5: note: in expansion of macro 'palSetPadMode'
     palSetPadMode(I2C1_SCL_BANK, I2C1_SCL, PAL_MODE_STM32_ALTERNATE_OPENDRAIN);
     ^~~~~~~~~~~~~
In file included from <command-line>:0:0:
drivers/arm/i2c_master.c: In function 'i2c_start':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:76:15: note: in expansion of macro 'I2C_DRIVER'
     i2cStart(&I2C_DRIVER, &i2cconfig);
               ^~~~~~~~~~
drivers/arm/i2c_master.c: In function 'i2c_transmit':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:82:15: note: in expansion of macro 'I2C_DRIVER'
     i2cStart(&I2C_DRIVER, &i2cconfig);
               ^~~~~~~~~~
drivers/arm/i2c_master.c: In function 'i2c_receive':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:89:15: note: in expansion of macro 'I2C_DRIVER'
     i2cStart(&I2C_DRIVER, &i2cconfig);
               ^~~~~~~~~~
drivers/arm/i2c_master.c: In function 'i2c_writeReg':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:96:15: note: in expansion of macro 'I2C_DRIVER'
     i2cStart(&I2C_DRIVER, &i2cconfig);
               ^~~~~~~~~~
drivers/arm/i2c_master.c: In function 'i2c_readReg':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:110:15: note: in expansion of macro 'I2C_DRIVER'
     i2cStart(&I2C_DRIVER, &i2cconfig);
               ^~~~~~~~~~
drivers/arm/i2c_master.c: In function 'i2c_stop':
./keyboards/lilith101/config.h:117:20: error: 'I2CD1' undeclared (first use in this function)
 #define I2C_DRIVER I2CD1
                    ^
drivers/arm/i2c_master.c:115:32: note: in expansion of macro 'I2C_DRIVER'
 void i2c_stop(void) { i2cStop(&I2C_DRIVER); }
                                ^~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:377: recipe for target '.build/obj_lilith101_default/i2c_master.o' failed
make[1]: *** [.build/obj_lilith101_default/i2c_master.o] Error 1
Makefile:560: recipe for target 'lilith101:default' failed
make: *** [lilith101:default] Error 1
tnf@tnf-VirtualBox:~/Desktop/github/qmk_firmware$ Make finished with errors
```

Default values for OPMODE_I2C and STD_DUTY_CYCLE are missing (that's why they are set in the config file), and USE_I2CV1 should be defined by the driver.

This is the furthest i could get with my skills.
Functionality is already supported in Chibios.
Hopefully the community can implement this properly since it will open up the functionality of QMK even further:)
I can provide hardware testing since it is already build and used daily.

yulei, 2019-12-09:
you can move #define USE_I2CV1 from mcuconf.h to config.h which could bypass this issue.
https://github.com/qmk/qmk_firmware/blob/master/keyboards/matrix/noah/config.h

And after pr #7526 was merged, you can add the KINETIS  support easier.

drashna, 2019-12-09:
Also need this for the input club keyboards, as they use a custom i2c implementation but need the `i2c_master` driver support to switch off of uGFX

St0RM53, 2019-12-09:
> 
> 
> you can move #define USE_I2CV1 from mcuconf.h to config.h which could bypass this issue.
> https://github.com/qmk/qmk_firmware/blob/master/keyboards/matrix/noah/config.h
> 
> And after pr #7526 was merged, you can add the KINETIS support easier.

It was already on mcuconf i just copied and pasted on the wrong field while writing the post. I edited it, sorry.

zvecr, 2019-12-09:
There is a small amount of change required outside of `USE_I2CV1` to get this working, mostly due to the fact the ChibiOS-contrib hal port is a tad inconsistent. I've opened #7591 to cover the changes, but it has a few dependencies to get in first.

St0RM53, 2019-12-10:
Thanks a lot @zvecr. I will wait until it's ready and then test :)

laughingrice, 2020-04-04:
Not sure if it helps, I merged the changes into master and managed to compile both my code and handwired/onekey on the teensy 3.2 with the i2c_scanner keymap, but no i2c device is found (error -1), I'm guessing that is the first step to get any I2C device working.

Will be happy for any pointers/examples/missing patches. Not sure if the problem is the I2C code or if it's using non-standard I2C pins

PR #7591 (2020-07-25): Enable OLED support for Teensy 3.2/LC

## Description

Another situation where the ChibiOS hal does a poor job of abstraction, where I2Cv1 has a different config structure.

**Note:** Teensy LC do not have enough memory to run a 128x32 OLED in the codebases current state.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #7574

## Checklist

- [x] My code follows the code style of this project.
- [ [... cut]
closed 2020-07-25 by tzarc