qmk/qmk_firmware #7575: ARM WS2812 driver not implemented for KINETIS MCU's (i.e. Teensy 3.2/3.5/3.6/LC)

St0RM53, 2019-12-09:
The WS2812 driver used for RGB Matrix functionality does not support KINETIS ARM microcontrollers.

## Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I have been waiting to implement this functionality on my finished keyboard i made about 3 months ago. This has been discussed on the discord server and while some process was made, the driver has no proper definitions to compile and work for KINETIS boards. 

Setting the following in:

**config.h**
```
#define RGB_DI_PIN E1
#define DRIVER_LED_TOTAL 101
```

will result in the following compilation output:

```
tnf@tnf-VirtualBox:~/Desktop/github/qmk_firmware$ make lilith101:default
QMK Firmware 0.7.97
Making lilith101 with keymap default

arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  34376	      0	  34376	   8648	.build/lilith101_default.hex

Compiling: keyboards/lilith101/led.c                                                                [OK]
Compiling: keyboards/lilith101/lilith101.c                                                          [OK]
Compiling: keyboards/lilith101/keymaps/default/keymap.c                                             [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgb_matrix.c                                                                     [OK]
Compiling: quantum/rgb_matrix_drivers.c                                                             [OK]
Compiling: drivers/arm/ws2812.c                                                                    drivers/arm/ws2812.c:12:10: error: #error ("NOP_FUDGE configuration required")
 #        error("NOP_FUDGE configuration required")
          ^~~~~
drivers/arm/ws2812.c: In function 'sendByte':
drivers/arm/ws2812.c:18:25: error: 'STM32_SYSCLK' undeclared (first use in this function)
 #define CYCLES_PER_SEC (STM32_SYSCLK / NUMBER_NOPS * NOP_FUDGE)
                         ^
drivers/arm/ws2812.c:20:36: note: in expansion of macro 'CYCLES_PER_SEC'
 #define NS_PER_CYCLE (NS_PER_SEC / CYCLES_PER_SEC)
                                    ^~~~~~~~~~~~~~
drivers/arm/ws2812.c:21:32: note: in expansion of macro 'NS_PER_CYCLE'
 #define NS_TO_CYCLES(n) ((n) / NS_PER_CYCLE)
                                ^~~~~~~~~~~~
drivers/arm/ws2812.c:25:29: note: in expansion of macro 'NS_TO_CYCLES'
         for (int i = 0; i < NS_TO_CYCLES(x); i++) { \
                             ^~~~~~~~~~~~
drivers/arm/ws2812.c:56:13: note: in expansion of macro 'wait_ns'
             wait_ns(T1H);
             ^~~~~~~
drivers/arm/ws2812.c:18:25: note: each undeclared identifier is reported only once for each function it appears in
 #define CYCLES_PER_SEC (STM32_SYSCLK / NUMBER_NOPS * NOP_FUDGE)
                         ^
drivers/arm/ws2812.c:20:36: note: in expansion of macro 'CYCLES_PER_SEC'
 #define NS_PER_CYCLE (NS_PER_SEC / CYCLES_PER_SEC)
                                    ^~~~~~~~~~~~~~
drivers/arm/ws2812.c:21:32: note: in expansion of macro 'NS_PER_CYCLE'
 #define NS_TO_CYCLES(n) ((n) / NS_PER_CYCLE)
                                ^~~~~~~~~~~~
drivers/arm/ws2812.c:25:29: note: in expansion of macro 'NS_TO_CYCLES'
         for (int i = 0; i < NS_TO_CYCLES(x); i++) { \
                             ^~~~~~~~~~~~
drivers/arm/ws2812.c:56:13: note: in expansion of macro 'wait_ns'
             wait_ns(T1H);
             ^~~~~~~
drivers/arm/ws2812.c: In function 'ws2812_setleds':
drivers/arm/ws2812.c:18:25: error: 'STM32_SYSCLK' undeclared (first use in this function)
 #define CYCLES_PER_SEC (STM32_SYSCLK / NUMBER_NOPS * NOP_FUDGE)
                         ^
drivers/arm/ws2812.c:20:36: note: in expansion of macro 'CYCLES_PER_SEC'
 #define NS_PER_CYCLE (NS_PER_SEC / CYCLES_PER_SEC)
                                    ^~~~~~~~~~~~~~
drivers/arm/ws2812.c:21:32: note: in expansion of macro 'NS_PER_CYCLE'
 #define NS_TO_CYCLES(n) ((n) / NS_PER_CYCLE)
                                ^~~~~~~~~~~~
drivers/arm/ws2812.c:25:29: note: in expansion of macro 'NS_TO_CYCLES'
         for (int i = 0; i < NS_TO_CYCLES(x); i++) { \
                             ^~~~~~~~~~~~
drivers/arm/ws2812.c:92:5: note: in expansion of macro 'wait_ns'
     wait_ns(RES);
     ^~~~~~~
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:377: recipe for target '.build/obj_lilith101_default/ws2812.o' failed
make[1]: *** [.build/obj_lilith101_default/ws2812.o] Error 1
Makefile:560: recipe for target 'lilith101:default' failed
make: *** [lilith101:default] Error 1
tnf@tnf-VirtualBox:~/Desktop/github/qmk_firmware$ Make finished with errors

```

Since it doesn't compile i can't verify if the E1 pin is defined (since in the past not all of the teensy 3.2 pins were defined). In that case this should be added in the config file:
#define E1 PAL_LINE(GPIOE, 1)

This is the furthest i could get with my skills.
Hopefully the community can implement this properly since it will open up the functionality of QMK even further:)
I can provide hardware testing since it is already build and used daily.

fauxpark, 2019-12-09:
For the Kinetis chips, the define appears to be `KINETIS_SYSCLK_FREQUENCY` rather than `STM32_SYSCLK`. `CYCLES_PER_SEC` will need to be made aware of this, and the NOP fudge factor(s) found for `K20x` and `KL2x`. Apart from that I don't see any other STM32-oriented code in the driver (though my eyes are not trained for ChibiOS).

St0RM53, 2019-12-09:
Ok i can put this then should compile. However i don't know what the proper NOP fudge factor value for K20x is, or what it actually is.
```
#define STM32_SYSCLK KINETIS_SYSCLK_FREQUENCY
#define NOP_FUDGE 0.4
```

closed 2022-08-03 by tzarc