qmk/qmk_firmware #7580: [Bug] Can't build USB-USB converter for Proton-C

dmelliot, 2019-12-09:
Building the USB-USB converter for Proton-C fails.

## Describe the Bug
I am trying to build the Pro-Micro based USB-USB converter code but for the Proton-C (as I need the additional storage space - see #7285) but it is failing.

The [docs](https://docs.qmk.fm/#/proton_c_conversion?id=converting-a-board-to-use-the-proton-c) suggest any Pro-Micro board can be converted to use the Proton-C, but perhaps the USB-USB converter with it's HID code is an exception?

Please let me know if there is any chance this might work in the future as I bought a couple of Proton-C boards specifically for this.

Thanks,
Duncan

## System Information

 - Keyboard: USB-USB converter (Pro-Micro)
 - Operating system: Ubuntu 18.04 (but building on QMK docker image)
 - AVR GCC version: 4.9.2 

 - ARM GCC version: 6.3.1

 - QMK Firmware version: 0.7.38

 - Any keyboard related software installed? None

## Additional Context
```
root@16787c8fc158:/qmk_firmware# make converter/usb_usb/pro_micro:default CTPC=yes
QMK Firmware 0.7.38
Making converter/usb_usb/pro_micro with keymap default

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/converter/usb_usb/custom_matrix.cpp                                           In file included from tmk_core/protocol/usb_hid/USB_Host_Shield_2.0/version_helper.h:24:0,
                 from tmk_core/protocol/usb_hid/USB_Host_Shield_2.0/settings.h:81,
                 from tmk_core/protocol/usb_hid/USB_Host_Shield_2.0/Usb.h:27,
                 from keyboards/converter/usb_usb/custom_matrix.cpp:22:
tmk_core/protocol/usb_hid/arduino-1.0.1/cores/arduino/Arduino.h:8:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:377: recipe for target '.build/obj_converter_usb_usb_pro_micro_default_proton_c/custom_matrix.o' failed
make[1]: *** [.build/obj_converter_usb_usb_pro_micro_default_proton_c/custom_matrix.o] Error 1
Makefile:555: recipe for target 'converter/usb_usb/pro_micro:default' failed
make: *** [converter/usb_usb/pro_micro:default] Error 1
Make finished with errors

```

fauxpark, 2019-12-09:
Unfortunately, the USB-USB converter (probably most if not all of the x-to-USB converters) is very AVR-specific, as it requires code that comes from this project: https://github.com/felis/USB_Host_Shield_2.0. The Proton C, being an Arm-based board, simply will not understand the AVRisms of it, so it's likely not possible short of rewriting the whole thing for STM32. Sorry :(

yulei, 2019-12-09:
usb host shield lib supports stm32f4 through the st’s HAL lib。I have ported to chibios, also tested against an custom hardware. But this lib uses hard coded  configuration and hardware specific codes, it was hard to integrated into qmk(also potential license conflict?). If you really want to use it with proton C, you’d better to fork the official lib, grep all the stm32f4 related code, replace them as needed.

dmelliot, 2019-12-10:
Thanks both of you for the responses.  Sounds like it would be possible, but there is potentially a fair bit of work to get there...

I came across this:
https://github.com/MichaelTien8901/STM32-with-Arduino-USB-Host-Shield
which initially sounded promising, but I'm not sure what is actually being documented there (and to which lib the documented updates would be applied).

Anyway, taking a step back, is there an easier solution to this problem (i.e. running out of flash space on a Pro-Micro when building the USB-USB converter)?

Would a AT90USB1286 based board (e.g. a Teensy++2.0) work with QMK and the USB HID code?

fauxpark, 2019-12-10:
It seems to compile for me, by copying the `pro_micro` directory & files to `teensy_pp`, setting `MCU` and `BOOTLOADER` accordingly, and adding a config.h with `#define CORE_TEENSY` (the USB HID code wants that). I don't have a Teensy to test with, but looks promising 👍

And on avr-gcc 8.3.0 I get: `26168/130048 (20%, 103880 bytes free)`

dmelliot, 2019-12-10:
Thanks @fauxpark that looks promising.
I'll source a Teensy and give it a shot (although I think it will need converting to 3.3v too).

closed 2022-08-03 by tzarc