qmk/qmk_firmware #7585: possible [Bug] , tap-dance with macros

username11236, 2019-12-09:
## Describe the Bug

I have searched issues and could not find anything related to my problem. In case I have overlooked smth useful, please let me know.

I believe this to be  bug, however the error might be caused by the user (me)

I have one key, with a tapdance function(QUOT_LAYR). On tap this key permanentely switches my layer to _NUMBP. On hold it switches to the layer _NAVC temporarily. (and back after release)
On _NAVC I have arrow keys and a macro. (CUST_ALTTAB)
The macro registrers KC_ALT and KC_TAB on press, and releases KC_TAB on release.
The tapdance key QUOT_LAYR releases KC_ALT on release.

This enables me to alt-tab through multiple entries and use arrow keys as well.

The only downside i experience with this setup is that the macro only executes after the timeout of the tapdance function. The button is completely unresponsive before that, simple keycodes are registered correctly though.

I tried to achieve this functionality using an advanced tap dance function instead of the macro, with exactly the same effect.

Any help is greatly appreciated!

## System Information

 - Keyboard:  
Signum3.1, elite-c (pro micro)
   - Revision (if applicable):  
 - Operating system:
ubuntu for build, win81 for flash (atmel flip)
 - AVR GCC version: 
avr-gcc (GCC) 5.4.0
 - ARM GCC version: 
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
 - QMK Firmware version:
0.7.97
 - Any keyboard related software installed? 
No

## Additional Context
keymap:
TD(QUOT_LAYR) on _BASE; CUST_ALTTAB on _NAVC
https://pastebin.com/2kjMp1WE

Shinxis, 2020-08-17:
I have similar problem with interrupted Tap Dance and changing to layers with custom keycodes.

My goal is to have an otherwise normal space key on thumb, but with the exception that when pressed quickly in conjunction with another key (interrupted single hold), it would change layer and send the keycode from there. And of course change the layer back when the Tap Dance key is released. 

Now the desired action works only after the TAPPING_TERM expired, i.e. when the hold is NOT interrupted. If the hold is interrupted, custom keycodes act as if they're on the wrong layer; on base layer they get called instead of those on the top layer, and on the upper layer the first tap outputs nothing. Basic keycodes seem to work fine.

Getting this functionality to work would be invaluable considering the four thumb keys on an Ergodox, meaning four instantly accessible layers with zero keys sacrificed to layer changing, and minimal impact on the regular key behavior. The regular keycodes may have failsafe duplicates on the thumb layers if necessary.

Has anyone got an idea for a fix or a workaround? Some custom Tap Dance / timer code and matrix_scan_user ?

Heres part of the code I'm using:

void td_rspace_each(qk_tap_dance_state_t *state, void *user_data) {  // Right thumb space - Navi/ArrowKey layer
  if (state->count > 1) {
    state->count = 1;
    tap_code(KC_SPACE);
  }
}
void td_rspace_finished(qk_tap_dance_state_t *state, void *user_data) {
  td_state[TD_RSPACE] = cur_dance(state);
  switch (td_state[TD_RSPACE]) {                               // every tap dance has its own td_state value
    case SINGLE_TAP:
      if (layer_state_is(NAVI_LAYER) {layer_off(NAVI_LAYER);}  // reset if layer is activated from somewhere else
      else                           {tap_code(KC_SPACE);   }
      break;
    case SINGLE_HOLD:
      register_code(KC_SPACE);
      break;
    case SINGLE_HOLD_QUICK:                                    // state->count == 1 && state->pressed && state->interrupted
      layer_on(NAVI_LAYER);
      break;
    default:
      break;
  }
}
void td_rspace_reset(qk_tap_dance_state_t *state, void *user_data) {
  switch (td_state[TD_RSPACE]) { 
    case SINGLE_HOLD:
      unregister_code(KC_SPACE);
      break;
    case SINGLE_HOLD_QUICK:
      layer_off(NAVI_LAYER);
      break;
    default:
      break;
  }
  td_state[TD_RSPACE] = 0;
}

wheredoesyourmindgo, 2021-05-23:
@username11236  I had this exact same problem. For me the workaround is to track the state of the tap dance and check if it's INTERRUPTED in the macro in the switch-case block. If it is INTERRUPTED and not pressed I let the macro run and clear the tap dance state after that code block does run. See [my qmk source](https://github.com/wheredoesyourmindgo/qmk_firmware/blob/c99e11188c1c202d5b27ba2ccea85366e957204f/users/wheredoesyourmindgo/wheredoesyourmindgo.c#L532) for example.

wheredoesyourmindgo, 2021-05-23:
The underlying issue appears to be that the record event pressed property is falsy during the first key in process_record_user()  press when keycode is triggered before the tap dance tap term expires.

closed 2022-08-29 by zvecr