qmk/qmk_firmware #7599: [Bug] KC_MUTE does not trigger cleanup action for One Shot Layers

drashna, 2019-12-10:
## Describe the Bug

Hit a OSL key, hit `KC_MUTE` on that layer. 

This should clear the layer immediately, on press. But it does not. It waits for the timeout period. 

## System Information

 - Keyboard:  Ergodox EZ
   - Revision (if applicable):  
 - Operating system: MacOS, Windows 
 - AVR GCC version: 8.3.1
 - QMK Firmware version: 0.7.100

## Additional Context

Other media keys don't seem to trigger this behavior, just `KC_MUTE`

fauxpark, 2019-12-11:
Confirming this. What's more, `layer_state_set_*` is behaving oddly. I have mine set up to turn on the scroll lock LED when my Fn layer is active, but when I tap the OSL key, it flashes briefly (but the layer does seem to be active). If I then press a key, the LED lights up and stays lit up despite the oneshot now being completed.

drashna, 2019-12-13:
That's ... odd.  I can't confirm the `layer_state_set_*` behavior.  That works fine for me, but I'm using AVR, so that may be why. 

Also, there is a PR that fixes the `timer_elapsed` function on chibiOS, and that may be related to your issue?

fauxpark, 2019-12-13:
I'm on AVR too. Specifically, I tested on a Wasdat and a Satan.

My code looks like this:
```c
#define FX_FNMN OSL(_FN)

// ... keymap

bool led_update_user(led_t led_state) {
    return false;
}

layer_state_t layer_state_set_user(layer_state_t state) {
    writePin(B2, IS_LAYER_ON(_FN));

    return state;
}
```

Also tried it with `BL_TOGG`, which turns the backlight off when pressed, and turns it back on when released. Functions normally when the OSL is held, BL_TOGG tapped, then OSL released.

sigprof, 2020-04-11:
Looks like `KC_MUTE` is converted [here](https://github.com/qmk/qmk_firmware/blob/58a9c84d6bb22c7544231f60acace4a85d6f8dd2/quantum/keymap_common.c#L68) to an `ACT_USAGE` action which contains a HID usage code instead of a QMK keycode, and the [usage code for `AUDIO_MUTE`](https://github.com/qmk/qmk_firmware/blob/f4799481cddcc8713ad3bfb80e844acc0981af69/tmk_core/common/report.h#L63) happens to be `0xE2`; then [this code](https://github.com/qmk/qmk_firmware/blob/58a9c84d6bb22c7544231f60acace4a85d6f8dd2/tmk_core/common/action.c#L227) misinterprets that `0xE2` as a QMK keycode, which then passes the `IS_MOD` check. Other media keys do not have the problem, because `AUDIO_MUTE` happens to be the only HID usage code used by QMK which has the low byte in the `0xE0 ... 0xE7` range.

fauxpark, 2020-04-12:
So I guess that line could simply check `action.kind.id != ACT_USAGE` as well?

spidey3, 2020-04-17:
There are actually two separate issues here:
1. The `KC_MUTE` issue, which I think is easily fixed, as @fauxpark suggests.
2. A separate, larger issue, that QMK functions (implemented in `process_record_quantum()` or its constituents) will never reach this code, leaving the OSL active. For this, I suppose code could be added to  `process_record()` [here](https://github.com/qmk/qmk_firmware/blob/58a9c84d6bb22c7544231f60acace4a85d6f8dd2/tmk_core/common/action.c#L188) to do the cleanup in case `process_record_quantum()` return `false`.

spidey3, 2020-04-17:
Correction: There is a third related issue:
3. The same assumption that breaks `KC_MUTE` also breaks layer setting actions.

drashna, 2020-08-01:
Fixed by #8832

closed 2020-08-01 by drashna