qmk/qmk_firmware #7611: PS2_MOUSE_ENABLE = no enables PS2

nmw, 2019-12-12:
I copied a keyboard's directory "/keyboards/handwired/108key_trackpoint" and used it as a template for my handwired keyboard.

I don't have a PS/2 trackpoint, so in rules.mk I set:

PS2_MOUSE_ENABLE = no
PS2_USE_USART = no

under the comments that say "change yes to no to disable"

When I looked at the matrix debug, though, I eventually got a bunch of PS2 errors, which were probably tying up the CPU, causing my original problem of extreme lag, and missed keypresses.

after instead commenting out the line, it works correctly.

It looks like the code just checks if they're defined, and not if they're set to no.

drashna, 2019-12-13:
Ah, you're correct.  The code is checking `ifdef PS2_MOUSE_ENABLE` rather than if it's set to "yes".

closed 2019-12-13 by zvecr

drashna, 2019-12-13:
This should work as expected, now.

PR #7620 (2019-12-13): core: Fix build config in protocol.mk
Backports a fix from tmk_keyboard

Fixes #7611 

## Types of Changes

- [x] Core
- [x] Bugfix