qmk/qmk_firmware #7624: QMK flashing problem, memory not clearing

CormickTreacy, 2019-12-13:
When attempting to flash a recently purchased second hand board i recieve the error:

`>>> dfu-programmer.exe atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
>>> dfu-programmer.exe atmega32u4 flash "C:\Users\username\Downloads\qmk60.hex"
    Checking memory from 0x0 to 0x607F...  Not blank at 0x1.
    The target memory for the program is not blank.
    Use --force flag to override this error check.
>>> dfu-programmer.exe atmega32u4 reset`

I have tried everything i can think of so any help would be greatly appreciated...

drashna, 2020-01-18:
Retry, that fixes the issue in a lot of cases.

Magascorada, 2020-02-06:
hat doesn't fix the issue, YMDK96 NOT clearing memory on reset in order to flash the QMK hex file.

NapoleonBLVD, 2020-04-21:
Hi guys,

Same here. Using latest QMK Toolbox and drivers now but I get the same message when attempting to re-flash my Melody96. Any help/pointer to a solution would be great.

davidgoldenberg, 2020-05-10:
also experiencing this issue, retrying just repeats the errors.

tomwang2011, 2020-05-21:
also experiencing this error

germ, 2020-05-26:
Had this issue before, easiest solution was to just force a firmware over with a ISP flasher. After that normal flashing methods worked fine.

NapoleonBLVD, 2020-05-27:
> Had this issue before, easiest solution was to just force a firmware over with a ISP flasher. After that normal flashing methods worked fine.

Hi, germ. Thanks for the response. I saw this solution a while back in the QMK docs when I was searching around, but this solution requires having another micro-controller and knowledge in soldering (which I don't possess which is why I typically opt for hotswappable boards.

Are you also suggesting perhaps there's no way around forcing it? I was thinking maybe I could attempt force flashing via mingw64, instead of QMK Toolbox, just looking for time to update my copy of qmk. Would this be worth it/a potential alternative to ISP flashing?

If not, it sounds cheaper to just buy a new PCB (and pray it doesn't get screwed up again), instead of investing in a micro-controller and soldering tools. (very short term, but I don't see myself buying higher-end boards in the near future with the economy here in the Philippines)

germ, 2020-05-27:
Correct, if you have an Arduino handy it shouldn't be too bad. Check if you have a HackerSpace nearby or anyone you know that would be able to flash it for you! An alternative solution is to hot-air the 32u4 off and slap a fresh replacement on. That's actually my go-to these days instead of breaking out a ISP flasher when that happens.

I don't think that using Mingw instead of toolbox will make much of a difference. But once you get it flashed, you should be able to flash it normally!

NapoleonBLVD, 2020-05-27:
Darn, was really hoping a software-only solution would work. But I understand why it wouldn't make much of any difference since they both work off the same stuff.

I appreciate the response and I guess I'd better get to soldering, sourcing, and or flat out buying a new PCB (whichever is most accessible).

NapoleonBLVD, 2020-05-27:
@germ so I'm considering the hot-air and replace solution but since I've never done that before could you clarify these for me:

1. Hot air like "heat gun hot" or "blow dryer for hair hot"?
2. I have zero clue how the 32u4 is attached to the pcb but since you're saying "hot-air" I figure you're implying it's attached via really fine pins and a paste (no solder work involved)?
3. And would the process be just heat till I can pluck it out with a pair of tweezers; slap on the new one and call it a day? Am I missing a step like "apply thermal paste"? Not really sure what specific material is needed.
4. I've found an Atmega32u4 chips in aliexpress. Are the model numbers important to match beyond the Atmega32u4?

Apologies if I sound too much like a noob, and based on the line of questioning I think it's more likely I might just buy a new PCB altogether (especially if a regular hair dryer won't work which means procuring a heat gun AND a new mcu which might collectively be more expensive and riskier from my POV).

germ, 2020-05-27:
1. Heat gun hot, it's specialized soldering equipment!
2. This does make the assumption you're not using a promicro or anything, but a onboard MCU. You'd need to look at your board
3. That's the gist of it. You may need to flux and add some solder to a few dry pins after. But heat, lift off with tweezers, place new one, heat
4. Not horribly, just make sure it's a TQFP package if your board uses a TQFP packaged 32u4. 

It's riskier, but more fun. Might be worth posting on mechmarket to see if anyone in your country can do ISP flashing for you. Might be the cheapest overall

NapoleonBLVD, 2020-06-14:
@germ hey man!

So I couldn't find a heat gun, and my local mechboard community in the PH talked me into ISP Flashing instead. Long story short it worked like a charm.

Essentially bought a Pro Micro, a multi-tester to trace where the breakout pins connect to my MCU, and soldering equipment, and followed the steps in the QMK ISP Flashing Guide as carefully as I could.

@CormickTreacy @Magascorada @tomwang2011 hey guys I think this actually works. Maybe list this as solved if you try it out yourself? :D

germ, 2020-06-18:
Good to hear! At least now you know what to do next time you see something like this :)

closed 2020-09-09 by fauxpark

ememmagadia, 2021-02-26:
> @germ hey man!
>
> So I couldn't find a heat gun, and my local mechboard community in the PH talked me into ISP Flashing instead. Long story short it worked like a charm.
>
> Essentially bought a Pro [...]

hey bro, im from PH also, im recieving the same issue , and i think i need the same solution, where can i contact you ?

PR #10070 (2020-09-09): Use the force when flashing with dfu-programmer

## Description

Theoretically, this should get dfu-programmer to stop whinging about "Not blank at 0x1". Testers appreciated 👍

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #7624

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conve [... cut]