qmk/qmk_firmware #7650: [Feature Request] Support "half disconnected" status LEDs in split_common

snortwolf, 2019-12-16:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

I'm slowly in the process of updating my split keyboards to use the common split code.  Ideally, I would like to have an LED light up when the halves are disconnected without implementing it via a custom matrix.  I've played around with it a bit and ended up with this fork:

snortwolf/qmk_firmware@0adcde1a4e00654a59724fc590f4ad7c23a96fc6

The keyboard portion, not included in that tree, is something like this:

````
void matrix_report_kb(bool error) {
    writePin(SCROLL_LOCK_PIN, !error);
}
````

Is this something anyone else would be interested in?  Thoughts and suggestions welcome.

drashna, 2020-01-18:
That would be nice to have.

fauxpark, 2020-01-18:
We could add this to the `led_update_*()` handlers, perhaps 🤔

drashna, 2020-01-18:
@fauxpark ideally, I think we should have some sort of hook for the split code so that custom actions can be called when the connection is lost or reacquired.   Eg, like re-initing the rgb matrix, toggling an LED, show a status on an OLED, etc. 

While we could just have an LED, I think something that is more flexible would be best here.

closed 2020-05-17 by stale[bot]