qmk/qmk_firmware #7659: [Bug] `keyboard_post_init_*` never called

FranticRain, 2019-12-17:
## Describe the Bug

Code placed in the `keyboard_post_init_*` or really any `matrix_*`/`keyboard_*` function is never called when placed in a `keyboard_name.c` file.

For example, using a Proton C and the `handwired/onekey` example:

`onekey.h`:
```C
// ...
static uint8_t wasInitializationCalled = 0;
// ...
```

`onekey.c`:
```C
// ...
void keyboard_post_init_kb(void)
{
    wasInitializationCalled = 1;
    keyboard_post_init_user();
}
// ...
```

`keymap.c`:
```C
// ...
enum macroKeycodes {
    INIT_TEST = SAFE_RANGE,
};
// ...
const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] =
{
    [0] = {
        { INIT_TEST }
    },
};
// ...
bool process_record_user(uint16_t keycode, keyrecord_t* record)
{
    switch (keycode)
    {
        case INIT_TEST:
        {
            if (record->event.pressed)
            {
                if (wasInitializationCalled)
                {
                    SEND_STRING("Yes");
                }
                else
                {
                    SEND_STRING("No");
                }
            }
        } break;
    }

    return true;
}
// ...
```

I would expect this example to respond with "Yes" when I click the key. But it always responds "No."

## System Information

 - Keyboard:  Proton C
   - Revision (if applicable):  1
 - Operating system: macOS Catalina 10.15.1
 - AVR GCC version: 7.4.0
 - ARM GCC version: 7.3.1
 - QMK Firmware version: 0.7.101
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I'm curious if this is just a Proton C/ARM bug, as I've never seen it mentioned before.

zvecr, 2019-12-17:
The issue is going to be due to `static uint8_t wasInitializationCalled = 0;` in the header file. This will be creating an instance of the variable per compilation unit. Instead:

onekey.h:
```c
// ...
extern uint8_t wasInitializationCalled;
```

onekey.c:

```c
uint8_t wasInitializationCalled = 0;
// ...
void keyboard_post_init_kb(void)
{
    wasInitializationCalled = 1;
    keyboard_post_init_user();
}
```

Where there is one instance, then an extern to it.

FranticRain, 2019-12-17:
Perfect. Thank you.

closed 2019-12-17 by FranticRain