qmk/qmk_firmware #7660: [Bug] Error building default keymap Planck EZ

mcp292, 2019-12-17:
## Describe the Bug
I'm following the getting started guide and when I go to build my default keymap using `make planck/ez:default` I get this error:
```
Compiling: tmk_core/common/chibios/bootloader.c                                                    In file included from ./lib/chibios/os/common/ext/CMSIS/ST/STM32F3xx/stm32f303xc.h:168,
                 from ./lib/chibios/os/common/ext/CMSIS/ST/STM32F3xx/stm32f3xx.h:153,
                 from ./lib/chibios/os/common/startup/ARMCMx/devices/STM32F3xx/cmparams.h:72,
                 from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70,
                 from ./lib/chibios/os/rt/include/ch.h:81,
                 from tmk_core/common/chibios/bootloader.c:3:
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function 'enter_bootloader_mode_if_requested':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a
clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler
*/
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm'
statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler
*/
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_planck_ez_default/common/chibios/bootloader.o] Error 1
Make finished with errors
make: *** [Makefile:579: planck/ez:default] Error 1

```

## System Information

 - Keyboard:  Planck EZ
   - Revision (if applicable):  
 - Operating system: Arch Linux make planck/ez:default
 - AVR GCC version: same error with 8.3.0 and 9.2.0

 - ARM GCC version: 9.2.0

 - QMK Firmware version: 0.7.102

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

zvecr, 2019-12-17:
See #7006, (specifically <https://github.com/qmk/qmk_firmware/issues/7006#issuecomment-564208375> if you dont want to downgrade your arm-none-eabi-gcc version).

An update to ChibiOS is being worked on, so hopefully this wont be a long term issue.

mcp292, 2019-12-17:
Awesome thank you! Do you know the last working version of arm-non-eabi-gcc I should downgrade to?

closed 2019-12-17 by mcp292