qmk/qmk_firmware #7669: RGBLIGHT default effect list

yiancar, 2019-12-18:
Hello all,

Creating this issue to further discuss the default effect list. This is something that I have had in mind for quite some time. In my experience features can creep to be enabled by default where no one is using them.

QMK in general has done very well to keep features disabled by default but in my opinion, the RGB list is getting a bit out of hand:)

As of this moment defining `RGBLIGHT_ANIMATIONS` will enable all of the RGB animations. As this is a bloat to firmware size, I propose this list does not include super heavy animations and letting those be enabled individually.

Some topics for discussion:
- Is this methodology acceptable?
- What is a super heavy animation?
- Should we create a review guide for new animations?

yiancar, 2019-12-18:
My opinion is:
- Yes I like this methodology
- One that uses a complex (either memory or computationally) algorithm (I am looking at you floating point)
- Yes, so new contributors can adjust their expectations. Hopefully this will drive more efficient code

drashna, 2019-12-18:
Modes that could be removed from the default
* RGBLIGHT_MODE_RGB_TEST - Cycles through Red, Blue, and Green.  Useful for testing LED channels. Not much else. 
* RGBLIGHT_MODE_ALTERNATING - Lights up one half of the strip and alternates between the two halves. Useful for testing splits. Not much else.
* RGBLIGHT_MODE_CHRISTMAS - Useful for Christmas. Not much else. 

We could move the first two to a secondary `RGBLIGHT_TEST_ANIMATIONS`, and have a `RGBLIGHT_ANIMATIONS_ALL` that enables everything, but not enabled by default?

yiancar, 2019-12-19:
`RGBLIGHT_TEST_ANIMATIONS` is something I like:)

tzarc, 2022-08-04:
Should be sorted already.