qmk/qmk_firmware #7673: [Bug] Crkbd won't boot using OLED_DRIVER, RGB_MATRIX and UNICODE

oprietop, 2019-12-18:
## Describe the Bug
Got a corne keyboard with heavy feature usage, the config is: https://github.com/oprietop/qmk_firmware/tree/master/keyboards/crkbd/keymaps/oprietop  
It used to work months ago but I recently synced my fork to master and now it will compile but won't even boot.  To make it boot with the current master i must disable one of OLED_DRIVER_ENABLE, RGB_MATRIX_ENABLE or UNICODE_ENABLE.
After some trial an error the last tagged version it works is 0.7.50 and the culprit is https://github.com/qmk/qmk_firmware/commit/947e61eaebf168c0c60be34914141cff9a69e599  
Reverting that one in the current master makes the the corne boot.

## System Information

 - Keyboard:  crkbd
   - Revision (if applicable):  
 - Operating system: Linux, Windows 10
 - AVR GCC version: 5.4.0
 - ARM GCC version: 15:6.3.1+svn253039-1build1
 - QMK Firmware version: 0.7.102

## Additional Context
PMd Drashna on reddit (thanks man!) and he already pinpointed a possible memory problem, maybe i'm stressing too much the device.

dsanchezseco, 2020-01-09:
had the same with mine, i'd even changed the elite-c in one of the halves (with no improvement) before noticing that disabling the oled fixed the issue.

In my case the problem was that one of the halves was really slowed and the keys had to be pressed for a second before the key was sent but if using more features maybe it would have being fully unresponsive as well.

drashna, 2020-01-18:
@dsanchezseco are you using `oled_write` or `oled_write_P`?  It makes a difference, as the first writes to memory, and the later one uses program flash instead.   

Also, it may be worth seeing if reverting the above commit helps, too

dsanchezseco, 2020-01-19:
@drashna yes, i'm using `oled_write_ln_p` in my code. Should i be using the other? I'll check the above commit when i have some time.

Thanks!

dsanchezseco, 2020-01-21:
nope, same result with the `947e61e` and with the `oled_write_ln` the left side remains (mostly) unresponsive. The keys have to remain pressed for about a second before the keypress is sent to the computer.

The weirdness is that the right side behaves normal.

keyboard-magpie, 2022-08-03:
Believe this should be remedied with more recent code, closing for now.

closed 2022-08-03 by keyboard-magpie