qmk/qmk_firmware #7692: [Bug] LM(0, MOD_LGUI) Not Working As Expected

macintacos, 2019-12-21:
For whatever reason, I can't get `LM(0, MOD_LGUI)` to function properly. My `keymap.c` is below.

## Describe the Bug

The gist of what I'm trying to accomplish is that I want to be able to drop to the `0` layer at will to execute arbitrary `LGUI(KC_whatever)` shortcuts, no matter what layer I'm in (which is why the `LM` shortcut is defined on every layer except `0`). The way it works now, holding the key while in another layer _does_ register as the correct modifier being held down, but for whatever reason when I hit another key (`LGUI(KC_O)` for example), nothing happens. Really unsure why, since in theory it should be working.

If it makes any difference, I used the QMK online configurator to compile/generate everything (I have not gotten into modifying any code directly outside of the online editor).

Any ideas?

## System Information

 - Keyboard:  `dztech/dz60rgb_wkl/v1`
 - Operating system: macOS 10.15.2
 - AVR GCC version: 
```
avr-gcc (GCC) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
- ARM GCC version: 
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [X] Karabiner

## Additional Context

`keymap.c`:
```
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
		[0] = LAYOUT_HHKB(LCA_T(KC_GRV), KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSLS, TG(3), LCAG_T(KC_TAB), KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSPC, LCTL_T(KC_ESC), KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, TG(1), KC_NO, KC_LALT, KC_LGUI, KC_SPC, KC_RGUI, KC_RALT, KC_NO),
		[1] = LAYOUT_HHKB(KC_NO, KC_NO, KC_NO, KC_NO, LGUI(KC_RGHT), KC_NO, LGUI(KC_LEFT), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TAB, KC_NO, LALT(KC_RGHT), LALT(KC_RGHT), SGUI(KC_R), KC_NO, KC_NO, LGUI(KC_Z), TO(0), KC_NO, LGUI(KC_V), KC_NO, KC_NO, KC_BSPC, TO(0), LGUI(KC_RGHT), KC_NO, KC_NO, KC_NO, KC_NO, KC_LEFT, KC_DOWN, KC_UP, KC_RGHT, KC_NO, KC_NO, KC_ENT, KC_LSFT, KC_NO, KC_DEL, KC_NO, TG(2), LALT(KC_LEFT), KC_NO, KC_NO, KC_NO, KC_NO, LGUI(KC_F), KC_RSFT, KC_NO, KC_NO, KC_LCTL, LM(0, MOD_LGUI), KC_NO, LM(0, MOD_LGUI), KC_RALT, KC_NO),
		[2] = LAYOUT_HHKB(RESET, KC_NO, KC_NO, KC_NO, SGUI(KC_LEFT), KC_NO, SGUI(KC_RGHT), KC_NO, KC_NO, KC_NO, SGUI(KC_LEFT), KC_NO, KC_NO, KC_NO, KC_TRNS, KC_TAB, KC_NO, LSFT(LALT(KC_RIGHT)), LSFT(LALT(KC_RIGHT)), KC_NO, KC_NO, LGUI(KC_C), KC_TRNS, KC_TRNS, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, TO(0), LGUI(KC_A), KC_NO, KC_NO, KC_NO, KC_NO, RSFT(KC_LEFT), RSFT(KC_DOWN), RSFT(KC_UP), RSFT(KC_RGHT), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, LSFT(LALT(KC_RIGHT)), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, LM(0, MOD_LGUI), KC_NO, LM(0, MOD_LGUI), KC_NO, KC_NO),
		[3] = LAYOUT_HHKB(KC_NO, LCA(KC_U), LCA(KC_I), LCA(KC_J), LCA(KC_K), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, LCA(KC_E), KC_NO, LCA(KC_T), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(0), KC_NO, KC_NO, LCA(KC_D), LCA(KC_F), LCA(KC_G), LCA(KC_LEFT), LCA(KC_DOWN), LCA(KC_UP), LCA(KC_RGHT), KC_NO, KC_NO, LCA(KC_ENT), KC_NO, KC_NO, KC_NO, LCA(KC_C), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, LCAG(KC_LEFT), LM(0, MOD_LGUI), KC_NO, LM(0, MOD_LGUI), LCAG(KC_RGHT), KC_NO)};
```

ridingqwerty, 2019-12-21:
This is actually expected behavior:

https://beta.docs.qmk.fm/detailed-guides/keymap#keymap-and-layers

The way the layer stack in QMK works, you generally cannot use layer movement keys to move into a lower layer, only a higher one.

It may be possible to achieve this with a custom macro, but not LM unfortunately.

macintacos, 2019-12-21:
_sad trombone_

Thank you @ridingqwerty (fantastic username btw), I'll figure out some other way around this. Thanks for pointing that doc out! I figured it'd be mentioned somewhere but couldn't find it. I'll close this out; happy holidays!

closed 2019-12-21 by macintacos