qmk/qmk_firmware #7723: [Question][Feature Request] Neo-Style Layer Modifiers

RichardSchorrig, 2019-12-26:
Hi there,
I was wondering if there is a way to implement Neo2 Layer switching: Holding one key momentary toggles the layer, and pressing one other key would then permanently switch to the layer.

I've tried this with combos, this works fine for the caps lock (both left and right shift keys), but it does not work with the other layers (symbols, numpad and so on). What I've done: I have MO(2) and MO(3) keys, and pressing both should switch to layer 3, but the layer does not stick.

Is there a way I did not see or is this unimplemented?

noroadsleft, 2019-12-27:
I'm not sure it's quite what you're after, but:

- `MO(2)` when held switches to layer 2
- `TO(3)` when *tapped* switches to layer 3 and stays on layer 3

Note that once you're in layer 3, layer 2 will be inaccessible, unless layer 3 has some keycode that will disable layer 3, such as `TO(0)`.

https://docs.qmk.fm/#/feature_advanced_keycodes?id=switching-and-toggling-layers

RichardSchorrig, 2019-12-27:
Thanks for your reply, but unfortunately this is not what I want. 
I already did a similar thing, where in layer 2, there is a TO(3)[ = key 2]; but my goal is to either press key 1 and then key 2 or the other way around. Also, layer 2 and 3 are essentially the same (except that TO(3)).

noroadsleft, 2019-12-28:
So you're trying to implement https://www.neo-layout.org/ ? Forgive me, as I've never heard of Neo layout so I'm not familiar with how it works, especially the layer switching.

But the target is:

Tap Key1, then tap Key2 -> enable and lock layer 3
Tap Key2, then tap Key1 -> enable and lock layer 3

Am I getting closer?

RichardSchorrig, 2019-12-28:
Not quite:
Hold Key1, tap Key2 -> lock layer 2
Hold Key2, tap Key1 -> lock layer 2
This combination should also unlock layer 2.

Hold Key1 OR Key2 -> toggle layer 2
Bonus: if layer 2 is locked, holding either Key 1 or 2 should momentary return to layer 0.

I already have a Neo2 layout on my EZ-Ergodox which I am satisfied with (created with the configurator under www.ergodox-ez.com, which is here:
https://configure.ergodox-ez.com/ergodox-ez/layouts/orxrQ/ABarB/0

But I am currently playing around in QMK to see how I can improve stuff.

username11236, 2020-01-08:
This should do it:

lay1 = standard layer
lay2 = symbol layer
lay3 = copy of lay2  # locked version
lay4 = copy of lay1

on lay1
button1 = MO(lay2)
button2 = MO(lay2)

on lay2
button1 = TO(lay3)
button2 = TO(lay3)

on lay3
button1 = MO(lay4)
button2 = MO(lay4)

on lay4 
button1 = TO(lay1)
button2 = TO(lay1)

closed 2020-05-07 by stale[bot]