qmk/qmk_firmware #7738: "getting started" instructions do not produce working artifact (Linux/Manjaro 18.1.5)

ammkrn, 2019-12-29:
Following the steps in the "Getting Started" guide on Manjaro 18.1.5, stock bashrc fails to produce working software.

Steps to reproduce:

+ update all installed packages via `pacman -Syu`
+ git clone --recurse-submodules https://github.com/qmk/qmk_firmware.git
+ cd qmk_firmware
+ util/qmk_install.sh
+ make clueboard/66/rev3:default

fails with error :
```
Making clueboard/66/rev3 with keymap default

avr-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/clueboard/66/rev3/rev3.c                                                      In file included from ./keyboards/clueboard/66/rev3/config.h:3,
                 from <command-line>:
quantum/config_common.h:29:18: fatal error: avr/io.h: No such file or directory
 #        include <avr/io.h>
                  ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_clueboard_66_rev3_default/keyboards/clueboard/66/rev3/rev3.o] Error 1
Make finished with errors
make: *** [Makefile:579: clueboard/66/rev3:default] Error 1
```

CurryFurious, 2019-12-29:
I think I had a similar error initially, could you try running it with sudo? If it works with that, you should setup udev rules so you wont have to run it as sudo: https://beta.docs.qmk.fm/faq/faq_build#linux-udev-rules

ammkrn, 2019-12-29:
`sudo make clueboard/66/rev3:default` produces the same error, unfortunately.

CurryFurious, 2019-12-31:
If you run `pacman -Qe | grep avr`, Do you see
```
avr-gcc 9.2.0-1
avr-libc 2.0.0-3
avrdude 1:6.3-6
```
(Versions wont matter as much)

ammkrn, 2020-01-01:
No, it only found avr-gcc. When I individually installed avrdude and avr-libc via pacman that solved the problem. I'm not sure why that is, since I both ran the install script in qmk_firmware/utils and also tried to install the depends in the script manually via pacman, but as a list (`sudo pacman -S arm-none-eabi-binutils arm-none-eabi-gcc ...`).

Thank you very much for the help. If you think this is something that other Arch/Manjaro users might run into let me know and I'll make a PR with a note in the "getting started" section.

focusaurus, 2020-01-02:
@ammkrn I'm getting a series of similar errors on arch. I keep trying to install more packages but there are many seemingly related possibilities and I'm flying blind as to which one I might need next. Here's my last error after resolving several relating to `arm-none-*` and `avr*`.

```
make ergodox_infinity:default
QMK Firmware 0.7.106
Making ergodox_infinity with keymap default

arm-none-eabi-gcc (Arch Repository) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_infinity/matrix.c                                                     In file included from keyboards/ergodox_infinity/matrix.c:18:
/usr/lib/gcc/arm-none-eabi/9.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_ergodox_infinity_default/matrix.o] Error 1
Make finished with errors
make: *** [Makefile:579: ergodox_infinity:default] Error 1
```

CurryFurious, 2020-01-02:
@focusaurus Do you have these packages:
```
arm-none-eabi-binutils 2.33.1-1
arm-none-eabi-gcc-8.3.0-1
arm-none-eabi-newlib 3.1.0-2
```
arm-none-eabi-gcc version 9.2 might not work, you should downgrade to 8.3 or 7.3.

ammkrn, 2020-01-03:
@focusaurus 

@CurryFurious to the rescue again. I got a different error when trying to build `ergodox-infinity:default` (about chibios tripping -Werror by using a deprecated feature) and was able to fix it by pinning avr-none-eabi-gcc to 8.3.0-1. Try :
```
sudo pacman -U https://archive.archlinux.org/packages/a/arm-none-eabi-gcc/arm-none-eabi-gcc-8.3.0-1-x86_64.pkg.tar.xz
```

Here's a list of the versions I'm on if you want to check your toolchain against mine.
```
arm-none-eabi-binutils  2.33.1-1
arm-none-eabi-gcc 8.3.0-1
arm-none-eabi-newlib 3.1.0-2
avrdude 1:6.3-6
avr-binutils 2.33.1-1
avr-libc 2.0.0-3
avr-gcc 8.3.0-1
dfu-programmer 0.7.2-2
dfu-util 0.9-2
```

focusaurus, 2020-01-03:
I was initially going down the path of trying to install the right arch packages until I discovered/remembered the docker option which is preferable so I bailed and just used docker.

ammkrn, 2020-01-09:
Closing for now. If anyone runs across this after the fact, the fixes (for me) seemed to be :
https://github.com/qmk/qmk_firmware/issues/7738#issuecomment-570011445
and
https://github.com/qmk/qmk_firmware/issues/7738#issuecomment-570445117

closed 2020-01-09 by ammkrn

rjhilgefort, 2020-12-12:
I'm seeing this issue running Arch. I followed the setup steps in the "getting started" guide, and then I tried what @ammkrn recommended in the comment right above. I'm trying to build the `ergodox_infinity:default` keymap'

[image]

dther, 2020-12-21:
@rjhilgefort I was also getting this same error on an unrelated board, `handwired/sick_pad`, until I installed `arm-none-eabi-newlib 3.1.0-2`. It was the only one I was missing. I'm guessing that there might have been some undocumented dependency change somewhere?

fauxpark, 2020-12-21:
@dther the `qmk` package for Arch/Manjaro [lists some dependencies as optional](https://archlinux.org/packages/community/any/qmk/) (such as newlib), when they really should not be. Unfortunately though we do not control this package.

rjhilgefort, 2020-12-23:
Well, I think installing the optional dependencies did something. I don't see the above error anymore, but I plenty of other errors. I was able to build another another firmware just fine though (Preonic) so perhaps it's something specifically with the Ergodox Infinity files? In any case, I resorted to building with Docker and that seems to be working.