qmk/qmk_firmware #7747: [Question] MCU for new keyboard

igagis, 2019-12-31:
Hi,

I have a Drevo Gramr 84 key keyboard which I would like to modify. In particular I want to replace the4 PAUSE key with INSERT key.

I know that Drevo Gramr MCU is some weird stuff (see https://www.reddit.com/r/MechanicalKeyboards/comments/8tgs1k/gramr_84_help_getting_info_on_microcontroller/), so I thought that I could just replace the MCU with the one with which I could use QMK. I suppose that keys wiring in that keyboard is something very common to all keyboards, so I think it will not be very hard to just drop in another MCU.

So, my question is what would be the best/easiest MCU I could use for that purpose with QMK?

I have some experience with AVR (atmega8/32/64). Could you recommend what exact AVR MCU I should pick to start working with QMK?

Br,
Ivan

fauxpark, 2020-01-01:
The pinout for this MCU is going to be completely different to any AVR or Arm chip, so it won't be interchangeable in that way. The easiest thing would really be to make a replacement PCB designed for QMK from the ground up.

The most common MCU used in QMK is the ATmega32U4, but if you want more flash space and prefer to stick with AVR, there is also the AT90USB1286 (one of the Teensy models uses this).

On the Arm side, you have the STM32F303, used in the Planck rev6, Preonic rev3 and Proton C, as well as a few other boards. When the ChibiOS upgrade PR eventually lands there will likely be a few more chips added to the mix, too.

This page might be helpful: https://docs.qmk.fm/#/compatible_microcontrollers

igagis, 2020-01-01:
Thanks for info, I'll try to go with `ATmega32U4`.

Regarding the chip interchangeability I did not mean that it will be compatible pin-to-pin of course, but I think it should be possible to un-solder the current chip and solder the `ATmega32U4` to contact pads with flexible wires (I can do precise soldering). So, the question is only about wiring of the keys themselves and, possibly, key LEDs, if that wiring is compatible with QMK logic or not. How do you think, is that possible?

fauxpark, 2020-01-02:
It's unlikely the switch matrix is doing something crazy, so yeah, as long as you can trace the rows and columns you're good to go :)

igagis, 2020-01-02:
Thanks for help!

closed 2020-01-02 by igagis