qmk/qmk_firmware #7770: [Feature Request] Dedicated PIN to control RGB LED power

thinxer, 2020-01-03:
Just60 keyboard uses a dedicated pin to control the power of underglow LEDs. This is designed to save battery in BLE mode. Currently QMK supports only `RGB_DI_PIN` to control the data channel. I propose to add a new `RGB_ENABLE_PIN` configuration and sync the bit with `rgblight_config.enable`.  

Another keyboard with a similar design is [yd68](https://github.com/qmk/qmk_firmware/blob/3fd8f160c3b3394f2e06643347143e46d880aa69/keyboards/yd68/yd68.c#L18).

## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

A new `RGB_ENABLE_PIN` configuration which syncs with `rgblight_config.enable`.

thinxer, 2020-01-03:
This should be a small feature to add. Nevertheless, I see there are multiple RGB Matrix implementations, such as `avr/ws2812`, `avr/ws2812_i2c`, `arm/ws2812`, etc. I do not think it is a good idea to repeat the support for `RGB_ENABLE_PIN` in each implementation. One level up, the code in `quantum` directory do not seem to touch PINs, it is not good to add this feature here, either.

It is actually easy to support this feature by using `extern rgblight_config_t rgblight_config` and `process_record_user`. We could update the PIN on `RGB_TOGGLE` keypress. However, this could be fragile as  it relies on a variable not exposed by QMK Core.

drashna, 2020-01-18:
This could be done with a custom driver, much the way that the Ergodox EZ does.  Which is to say, it pulls the correct implementation for the ws2812 drives, and then modifies `rgblight_set` to be what it needs.

thinxer, 2020-01-18:
There are many ways to get it working, I'm just not sure if we should get this feature into QMK core. I could work on it if there are other keyboards sharing this design.

closed 2020-05-17 by stale[bot]