qmk/qmk_firmware #7789: Make can't find some keyboards [Bug]

levitanong, 2020-01-04:
When I try to `make dztech/dz65rgb:default`, I get:

```
QMK Firmware 0.7.106
make: *** No rule to make target 'dztech/dz65rgb:default'. Stop.
|
|  QMK's make format recently changed to use folder locations and colons:
|     make project_folder:keymap[:target]
|  Examples:
|     make planck/rev4:default:dfu
|     make planck:default
|
```
Even the second example: `make planck:default` yields the same error.

So far, I've tried most of the dztech keyboard:keymaps, and it's the same.

fauxpark, 2020-01-04:
There are two versions of these dztech boards - `dztech/dz65rgb/v1` etc. use an Arm MCU (STM32F303), and `v2` are AVR (ATmega32U4). They've been refactored recently so you have to know which version you have, and append the correct one to the make command.

levitanong, 2020-01-04:
That worked, thank you! Perhaps the readme should be updated then :) https://github.com/qmk/qmk_firmware/blob/master/keyboards/dztech/dz65rgb/readme.md

closed 2020-01-04 by levitanong

fauxpark, 2020-01-04:
Good point! I've added it to https://github.com/qmk/qmk_firmware/pull/7664 😃