qmk/qmk_firmware #7822: [Bug] json.decoder.JSONDecodeError

chiboreache, 2020-01-07:
## Describe the Bug

Can't create .C file from .json from http://qmkeyboard.cn/

## System Information

 - Keyboard:  dz60
  - Operating system: Archlinux
 - AVR GCC version: 9.2.0
 - QMK Firmware version: aur/qmk 0.0.15-2 (+0 0.00%) (Installed)

## Additional Context

```

↪ qmk json_keymap ./qmk60_4.json
<class 'json.decoder.JSONDecodeError'>
☒ Unexpected UTF-8 BOM (decode using utf-8-sig): line 1 column 1 (char 0)
Traceback (most recent call last):
File "/usr/lib/python3.8/site-packages/milc.py", line 564, in __call__
return self.__call__()
File "/usr/lib/python3.8/site-packages/milc.py", line 569, in __call__
return self._entrypoint(self)
File "/home/chibo/qmk_firmware/lib/python/qmk/cli/json/keymap.py", line 36, in json_keymap
user_keymap = json.load(fd)
File "/usr/lib/python3.8/json/__init__.py", line 293, in load
return loads(fp.read(),
File "/usr/lib/python3.8/json/__init__.py", line 337, in loads
raise JSONDecodeError("Unexpected UTF-8 BOM (decode using utf-8-sig)",
json.decoder.JSONDecodeError: Unexpected UTF-8 BOM (decode using utf-8-sig): line 1 column 1 (char 0)
↪

```

zvecr, 2020-01-07:
<http://qmkeyboard.cn/> isnt a compatible format, the json files in this context are the one <https://config.qmk.fm> uses.

chiboreache, 2020-01-07:
@zvecr that was not so obvious, but thanks anyway!

also there is no macro recording functionality, meh...

closed 2020-01-07 by chiboreache

chiboreache, 2020-01-08:
and I also can't make nested logic like `LT(1,TG(1))` which is meh `~_~`

chiboreache, 2020-01-08:
and fm json having zero semantic