qmk/qmk_firmware #7827: [Bug] [Unicode] getting hex numbers instead of symbols

chiboreache, 2020-01-08:
Maked a rule, compiled and flashed just fine, but when I press the button I get `30c4` instead of `ツ`

`[2] = LAYOUT(UC(0x30C4).....`

fauxpark, 2020-01-08:
Did you set up Unicode input on the host side? Unfortunately one of the drawbacks of this feature is that it is not exactly plug and play. See this section of the Unicode docs: https://docs.qmk.fm/#/feature_unicode?id=input-modes

chiboreache, 2020-01-08:
ok, get it

I added

```

void eeconfig_init_user(void) {
    set_unicode_input_mode(UC_LNX);
}

```

at the end of my code and I also mapped `UNICODE_MODE_LNX` to some key, but still getting the same result

chiboreache, 2020-01-08:
adding `#define UNICODE_SELECTED_MODES UC_LNX` into config.h - helped 㴠 ツ

wish you docs and example projects would be more informative and relevant :₽

closed 2020-01-08 by chiboreache

chiboreache, 2020-01-08:
but now I realised that it put its not instantly, but with every single char write and convert then - which is annoying as hell 㴠!㴠

drashna, 2020-01-08:
> it writes all symbols and convert them...

Yup, that's how the unicode input works.  Unfortunately, there isn't really a way around that, as the HID interface doesn't support unicode natively.

chiboreache, 2020-01-08:
what a shame, блядь...

could you tell which board can support it?

fauxpark, 2020-01-08:
None. This is a limitation of how keyboards work in general.

drashna, 2020-01-08:
It's not a matter of support for it. It's that the HID specs don't support it. 

Anything outside of the normal  Basically anything not on this list is not supported by the USB HID specs:
https://github.com/qmk/qmk_firmware/blob/8fa64568f01604bde0bb5eb1923e25483af03424/tmk_core/common/keycode.h#L214-L451

chiboreache, 2020-01-08:
Oh, so you mean that impossible, but what if I want custom unicode layout, so I need kind of OS on top of "weak" HID interface, right? Well maybe ARM boards can handle with it, huh?

drashna, 2020-01-08:
No, you'd have to implement something on the host system to manage it.   Or you'd have to use a custom endpoint .... with something on the host side to read and process it.

chiboreache, 2020-01-08:
Yeah, i mean the same, i need to make some firmware and install it or make some middle service, that would acts like helper

but, my holidays about the end, so i just make another US layout, which is mapped by OS to another layout, with a custom "shift" modifier, and rumble with it a little

as for symbols, i don't use it much, but it still annoying, only if xmodmap was powerful enough....meh 

anyway, thank you for your advice @drashna !