qmk/qmk_firmware #7837: [Feature Request] support for podman in util/docker_build.sh

joseph-sayler, 2020-01-09:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I was trying out util/docker_build.sh to build my keymap and noticed it did not support podman. It is a really easy fix to add support in the script. Not sure if this is of interest to anyone but me, just wanted to float the idea out there. This is my first time contributing so I thought I would post here first before submitting a PR.

Thanks!

zvecr, 2020-01-09:
Could you detail what support needs adding, like is this adding support for the non aliased podman form?

danielo515, 2020-01-25:
How is that script supposed to work? It always fail for me.
I have a much more simpler ones that always works

joseph-sayler, 2020-01-26:
Hey sorry I haven't responded sooner. Initially, I edited the provided
shell script and replaced the word docker with podman. Worked just fine. A
script change may not be necessary if the user has aliased podman as
docker. All docker arguments work with the podman command with no changes.

danielo515, 2020-01-26:
I spoke too soon. The script works fine, I was using an incorrect version of QMK.

closed 2020-05-25 by stale[bot]