qmk/qmk_firmware #7840: [Feature Request] implement "shift" functionality

chiboreache, 2020-01-09:
i wanna create complex keycode, just like in `XP(i, j)` i need possibility to make my own shift layer

```
instead:   7 &
i want:    ? 7
```

here is some workaround from @ridingqwerty

```
    case MY_7:
        if (record->event.pressed) {
            if(get_mods() & MOD_MASK_SHIFT)
            {
                clear_mods();
                tap_code(KC_7);
                set_mods(mod_state);
            } else {
                tap_code(KC_QUESTION);
            }
        }
        break;
```

it's great and it works, but not exactly suited for my purpose of completely redoing standard layout

ridingqwerty, 2020-01-09:
So, here's an example that should somewhat succinctly let you remap keys that expands on the above.

First, you need to add a custom keycode for each key you're changing like this; I'll be doing three in this example.  You would enumerate these just like other macro keycodes, but it's very important that they remain contiguous, e.g. no other keycodes interrupt their order:

```
enum custom_keycodes {
  MY_1 = SAFE_RANGE,
  MY_2,
  MY_3
};
```

Next, above process_record_user, you need to declare a multidimensional array of the keycodes to be used in this way -- the first group in `{}` is the unshifted chars, and the next is their corresponding shifted code.  Again, order is important here.

```
uint16_t alt_keymap[2][3] = {{KC_EXCLAIM, KC_A, KC_AMPR},{KC_1, KC_ASTR, KC_7}};
/* unshifted       shifted
 * !                    1
 * a                   *
 * &                   7
 */
```

Finally you need to tweak the snippet from above as follows:
```
    case MY_1 ... MY_3:
      if (record->event.pressed) {
        mod_state = get_mods() & MOD_MASK_SHIFT;
        clear_mods();
        if(mod_state) {
          tap_code16(alt_keymap[1][keycode - MY_1]);
        } else {
          tap_code16(alt_keymap[0][keycode - MY_1]);
        }
        set_mods(mod_state);
      }
      break;
```

This way, you only need one `process_record_user` "case", the only tricky part is managing and tracking each key and state and ensuring you carefully maintain their order.

ridingqwerty, 2020-01-09:
Alternatively, you can totally utilize the XP feature, bearing in mind that you go through the usual steps to enable unicode (`UNICODEMAP_ENABLE = yes` in rules.mk, setting the appropriate input method in QMK, and whatever needed host-side changes).

```
enum unicode_names {
    MY_AMPR,
    MY_7
};

// first 128 entries in the map can be used with the XP feature
const uint32_t PROGMEM unicode_map[] = {
    [MY_AMPR]  = 0x0026,  // &
    [MY_7] = 0x0037,  // 7
};

// creating a handy shortname to use in keymap; not strictly necessary
#define AMPR_7 XP(MY_AMPR, MY_7)
```

Now you can put `AMPR_7` directly into your keymap and it will behave as desired.

NOTE: this method is slower as there is a slight delay on the host side when entering unicode, so this may be untenable.

chiboreache, 2020-01-09:
desided to try second way first and get

```bash
Compiling: keyboards/dz60/keymaps/dz60_layout_mine/keymap.c                                        In file included from keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:1:
keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:45:16: error: implicit declaration of function 'XP'; did you mean 'X_P'? [-Werror=implicit-function-declaration]
45 | #define AMPR_7 XP(MY_AMPR, MY_7)
|                ^~
keyboards/dz60/dz60.h:33:91: note: in definition of macro 'LAYOUT'
33 |     { k30,  k31,   k32,   k33,  k34,  k35,   k36,  k37,   k38,  k39,   k3a,  k3b,  KC_NO, k3d,  k3e   }, \
|                                                                                           ^~~
keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:127:126: note: in expansion of macro 'AMPR_7'
127 | LSHIFT,XXXXXXX,  KC_Z,    KC_X,    KC_C,    KC_V,     KC_B,     KC_N,     KC_M,     KC_COMM,  KC_DOT,   KC_SLSH,  AMPR_7,  TT(LAT_UP),
|                                                                                                                   ^~~~~~

keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:45:16: error: initializer element is not constant
45 | #define AMPR_7 XP(MY_AMPR, MY_7)
|                ^~
keyboards/dz60/dz60.h:33:91: note: in definition of macro 'LAYOUT'
33 |     { k30,  k31,   k32,   k33,  k34,  k35,   k36,  k37,   k38,  k39,   k3a,  k3b,  KC_NO, k3d,  k3e   }, \
|                                                                                           ^~~
keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:127:126: note: in expansion of macro 'AMPR_7'
127 | LSHIFT,XXXXXXX,  KC_Z,    KC_X,    KC_C,    KC_V,     KC_B,     KC_N,     KC_M,     KC_COMM,  KC_DOT,   KC_SLSH,  AMPR_7,  TT(LAT_UP),
|                                                                                                                   ^~~~~~

keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:45:16: note: (near initialization for 'keymaps[0][3][13]')
45 | #define AMPR_7 XP(MY_AMPR, MY_7)
|                ^~
keyboards/dz60/dz60.h:33:91: note: in definition of macro 'LAYOUT'
33 |     { k30,  k31,   k32,   k33,  k34,  k35,   k36,  k37,   k38,  k39,   k3a,  k3b,  KC_NO, k3d,  k3e   }, \
|                                                                                           ^~~
keyboards/dz60/keymaps/dz60_layout_mine/keymap.c:127:126: note: in expansion of macro 'AMPR_7'
127 | LSHIFT,XXXXXXX,  KC_Z,    KC_X,    KC_C,    KC_V,     KC_B,     KC_N,     KC_M,     KC_COMM,  KC_DOT,   KC_SLSH,  AMPR_7,  TT(LAT_UP),
|                                                                                                                   ^~~~~~

cc1: all warnings being treated as errors
[ERRORS]
|
```

ridingqwerty, 2020-01-09:
Share a link to your code if possible; there's a couple very easy ways to trip up either method and I may not have been explicit enough.

ridingqwerty, 2020-01-09:
For those following along at home, issue was due to usage of `UNICODE` rather than `UNICODEMAP`, which is required for `XP()` function.

drashna, 2020-01-18:
I think you're lookin for something like #4795

closed 2020-05-17 by stale[bot]