qmk/qmk_firmware #7846: Unable to compile 1up60hse firmware in WSL [Bug]

tkissing, 2020-01-09:
## Describe the Bug

When trying to compile the default firmware for the 1up60hse I get an error during linking. Firmware for my other keyboards (dztech/dz65rgb/v1 and massdrop/alt) build fine. 

I can still build it on MacOS, so seems WSL specific.

## System Information

 - Keyboard:  1up60hse
   - Revision (if applicable):  
 - Operating system:
`Linux SPF15 4.4.0-18362-Microsoft #476-Microsoft Fri Nov 01 16:53:00 PST 2019 x86_64 x86_64 x86_64 GNU/Linux`
 - AVR GCC version: 
`avr-gcc (GCC) 4.8.2`
 - ARM GCC version: 
`arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2`
 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [x] AutoHotKey (in Windows, not in the WSL directly)
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

```
timok@SPF15:~/qmk_firmware$ make 1upkeyboards/1up60hse:default:bin
QMK Firmware 0.7.106
Making 1upkeyboards/1up60hse with keymap default and target bin

avr-gcc (GCC) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/1upkeyboards/1up60hse/1up60hse.c                                               [OK]
Compiling: keyboards/1upkeyboards/1up60hse/keymaps/default/keymap.c                                 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight.c                                                                       [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: quantum/backlight/backlight.c                                                            [OK]
Compiling: quantum/backlight/backlight_avr.c                                                        [OK]
Compiling: drivers/avr/ws2812.c                                                                     [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/avr/suspend.c                                                            [OK]
Compiling: tmk_core/common/avr/timer.c                                                              [OK]
Compiling: tmk_core/common/avr/bootloader.c                                                         [OK]
Assembling: tmk_core/common/avr/xprintf.S                                                           [OK]
Compiling: tmk_core/common/magic.c                                                                  [OK]
Compiling: tmk_core/common/command.c                                                                [OK]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: tmk_core/protocol/lufa/outputselect.c                                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/1upkeyboards_1up60hse_default.elf                                                   [ERRORS]
 |
 | In function ‘__vector_21’:
 | tmk_core/common/avr/timer.c:126:1: error: ‘_vector_21’ appears to be a misspelled interrupt handler [-Werror]
 |  ISR(TIMER_INTERRUPT_VECTOR, ISR_NOBLOCK) { timer_count++; }
 |  ^
 | lto1: all warnings being treated as errors
 | lto-wrapper: avr-gcc returned 1 exit status
 | /usr/lib/gcc/avr/4.8.2/../../../avr/bin/ld: lto-wrapper failed
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/1upkeyboards_1up60hse_default.elf] Error 1
make: *** [1upkeyboards/1up60hse:default:bin] Error 1
Make finished with errors
```

fauxpark, 2020-01-10:
4.8.2 is a pretty ancient GCC version to be using - try updating the Ubuntu installation WSL is running. For example Disco (19.04) provides `gcc-avr` 5.4.0, which is only slightly less ancient but is known to work.

tkissing, 2020-01-15:
Updating was quite a pain, but fixed the issue.
Maybe it would be good to have some check for minimum versions of critical build tools?

drashna, 2020-01-18:
You mean like `bin/qmk doctor`? :)

tkissing, 2020-01-28:
Now find a good place in build failure output to point that out and others running into the issue I faced might not even have to search GH issues for a solution ;)

(I know, documentation is hard, discoverability is harder)

closed 2022-08-03 by tzarc