qmk/qmk_firmware #7857: Compiling split keyboards does not work in MacOS

ZNielsen, 2020-01-10:
I use a multitude of split keyboards (lily58, crkbd, and a custom hand wired split board).  I recently picked up a Mac and I discovered that when compiling, all will go smoothly, but the programmed keyboard does not function properly; the non-usb half is unresponsive.  Compiling the same code on Windows Subsystem for Linux (WSL) works just fine.

Some tests I did:
- I transferred the .hex file from WSL to MacOS.  Using the QMK_TOOLBOX to flash the WSL compiled hex on MacOS worked fine.
- I compared the .hex files from WSL and MacOS.  They are very different.  I'm not sure if that's expected due to platform/compiler differences.
- The problem exists over all the split boards I have.  My custom board uses serial, while the lily58 uses I2C (I think?), so I think this problem exists with both communication methods.

Has this been encountered by anybody else?  Any ideas on a fix?

Thanks!

bcat, 2020-01-11:
I wonder if this is a more general problem with avr-gcc 8.3.0. I'm got a (more recent?) install of QMK on MSYS with that avr-gcc version, and after I just updated master and reflashed my Lily58, I have the same problem of the slave side not responding. I'll try and debug further.

bcat, 2020-01-11:
Turning off `LINK_TIME_OPTIMIZATION_ENABLE` in my `rules.mk` seems to the fix the problem. Some combination of LTO and 8.3.0 seems to be making my Lily58 slave side nonfunctional. I wonder if your problem is the same, and Windows vs. Mac is a red herring.

bcat, 2020-01-11:
@ZNielsen, are you perchange using LTO? If so, can you please patch in #7868 and see if it fixes this for you? (Though I'm not positive your problem is not unrelated, since you mentioned the crkbd, and that keeb seems to work for me with LTO enabled on latest master.)

ZNielsen, 2020-01-11:
I do believe I’m using link time optimization. I’ll give that a try on the boards I have at home. I can test the lily58 on Monday.

ZNielsen, 2020-01-13:
I was able to confirm that LTO was the problem.  Compiling and flashing without LTO, the lily58 behaved as expected.  Patching in `08cd99683` (#7868) fixed the problem with LTO enabled.

Should I change the title of this issue to make it more searchable?

tzarc, 2020-02-24:
Modified the title, closed it off as solved.

closed 2020-02-24 by tzarc