qmk/qmk_firmware #7866: No rule to make target 'myCustomKeymap'. Stop - Make finished with errors

pmc222, 2020-01-11:
Hi there,

I am building out my first keyboard, the Keebio Viterbi Rev2, and am running into issues I don't understand how to resolve when "building my first firmware" (following the instructions in https://docs.qmk.fm/#/newbs_building_firmware). 

1) I have the qmk_firmware directory chosen in MSYS2

2) I copied the keymaps/default folder and renamed it my github name, then saved the keymap I created as 'myCustomKeymap.c' in said folder (image attached)
[image]

3) I then attempted to execute the make command in MSYS2 MSYS:

make keebio/viterbi/rev2:myCustomKeymap (also tried make viterbi/rev2:myCustomKeymap)

Resulted in:
[image]

For kicks, I also executed:
make keebio/viterbi/rev2, which returned the following (sorry for the length):

Making keebio/viterbi/rev2 with keymap bakingpy                                  [OK]
Making keebio/viterbi/rev2 with keymap default                                                         [OK]
Making keebio/viterbi/rev2 with keymap default_ortho_5x14                                              [OK]
Making keebio/viterbi/rev2 with keymap drashna                                                         [OK]
Making keebio/viterbi/rev2 with keymap drashna_old                                                     [OK]
Making keebio/viterbi/rev2 with keymap dwallace                                                        [OK]
Making keebio/viterbi/rev2 with keymap fido                                                            [OK]
Making keebio/viterbi/rev2 with keymap mike808                                                         [OK]
Making keebio/viterbi/rev2 with keymap pmc222                                                          [ERRORS]
In file included from keyboards/keebio/viterbi/viterbi.h:8,
                 from keyboards/keebio/viterbi/keymaps/pmc222/keymap.c:1:
keyboards/keebio/viterbi/keymaps/pmc222/keymap.c:58:5: error: 'KC_Ins' undeclared here (not in a function); did you mean 'KC_I'?
     KC_Ins, KC_UNDS, KC_F1  , KC_F2  , KC_F3  , KC_F4  , KC_F5  ,     KC_F6  , KC_F7  , KC_F8  , KC_F9  , KC_F10 , KC_F11 , KC_F12 ,
     ^~~~~~
keyboards/keebio/viterbi/rev2/rev2.h:23:5: note: in definition of macro 'LAYOUT'
   { L00, L01, L02, L03, L04, L05, L06 }, \
     ^~~
keyboards/keebio/viterbi/keymaps/pmc222/keymap.c:57:14: note: in expansion of macro 'LAYOUT_ortho_5x14'
   [_RAISE] = LAYOUT_ortho_5x14(
              ^~~~~~~~~~~~~~~~~
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_keebio_viterbi_rev2_pmc222/keyboards/keebio/viterbi/keymaps/pmc222/keymap.o] Error 1
Making keebio/viterbi/rev2 with keymap yet-another-developer                                           [ERRORS]
layouts/community/ortho_5x14/yet-another-developer/keymap.c:13:35: error: macro parameters must be comma-separated
                                   L36,              R30,                               \
                                   ^~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:37:3: error: implicit declaration of function 'LAYOUT_ergodox_pretty_wrapper'; did you mean 'LAYOUT_ergodash_pretty_wrapper'? [-Werror=implicit-function-declaration]
   LAYOUT_ergodox_pretty_wrapper( \
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:47:56: note: in expansion of macro 'LAYOUT_ergodash_pretty_base'
 #define LAYOUT_ergodash_pretty_base_wrapper(...)       LAYOUT_ergodash_pretty_base(__VA_ARGS__)
                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:71:15: note: in expansion of macro 'LAYOUT_ergodash_pretty_base_wrapper'
   [_QWERTY] = LAYOUT_ergodash_pretty_base_wrapper(
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:42:34: error: 'EISU' undeclared here (not in a function); did you mean 'EIFR'?
       KC_LCTL, KC_LGUI, KC_LALT, EISU,                                                                                 KC_LEFT, KC_DOWN, KC_UP,   KC_RGHT  \
                                  ^~~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:47:56: note: in expansion of macro 'LAYOUT_ergodash_pretty_base'
 #define LAYOUT_ergodash_pretty_base_wrapper(...)       LAYOUT_ergodash_pretty_base(__VA_ARGS__)
                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
layouts/community/ortho_5x14/yet-another-developer/keymap.c:71:15: note: in expansion of macro 'LAYOUT_ergodash_pretty_base_wrapper'
   [_QWERTY] = LAYOUT_ergodash_pretty_base_wrapper(
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from tmk_core/common/action.h:23,
                 from quantum/keymap.h:23,
                 from quantum/quantum.h:30,
                 from keyboards/keebio/viterbi/viterbi.h:3,
                 from layouts/community/ortho_5x14/yet-another-developer/keymap.c:1:
[... 51 lines cut ...]
             ),
             ^
layouts/community/ortho_5x14/yet-another-developer/keymap.c:194:13: error: macro "LAYOUT_ergodash_pretty" passed 74 arguments, but takes just 70
             ),
             ^
layouts/community/ortho_5x14/yet-another-developer/keymap.c:205:13: error: macro "LAYOUT_ergodash_pretty" passed 76 arguments, but takes just 70
             ),
             ^
cc1.exe: all warnings being treated as errors
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_keebio_viterbi_rev2_yet-another-developer/layouts/community/ortho_5x14/yet-another-developer/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:579: keebio/viterbi/rev2] Error 1

I don't understand the error related to 
'KC_Ins' undeclared here (not in a function); did you mean 'KC_I'?
     KC_Ins, KC_UNDS, KC_F1  , KC_F2  , KC_F3  , KC_F4  , KC_F5  ,     KC_F6  , KC_F7  , KC_F8  , KC_F9  , KC_F10 , KC_F11 , KC_F12 ,

Here is a screenshot of the only place I have KC_Ins
[image]

I'm at a loss here. Everything was pretty straightforward up until this point. I also don't know where 
Making keebio/viterbi/rev2 with keymap yet-another-developer is coming from, as I don't see "yet-another-developer anywhere in the folder structure. Thanks for any help!

noroadsleft, 2020-01-11:
1. delete or rename `keyboards/keebio/viterbi/keymaps/pmc222/keymap.c`
2. rename `keyboards/keebio/viterbi/keymaps/pmc222/myCustomKeymap.c` to `keyboards/keebio/viterbi/keymaps/pmc222/keymap.c`

QMK's `make` configuration only builds files named `keymap.c`. The keymap part of the make command is the folder name. So your `make` command should be:

`make keebio/viterbi/rev2:pmc222`

and your keymap file is:

`keyboards/keebio/viterbi/keymaps/pmc222/keymap.c`

pmc222, 2020-01-11:
I figured it out. Stupid mistakes

closed 2020-01-11 by pmc222

noroadsleft, 2020-01-11:
Everyone starts somewhere. No worries!

pmc222, 2020-01-11:
Thanks for the quick reply, though! Appreciate the help :)

pmc222, 2020-01-11:
Even if you are a Ducks fan.... ;)

noroadsleft, 2020-01-11:
:duck: