qmk/qmk_firmware #7875: Switch between persistent layers

pmc222, 2020-01-12:
Read through tons of documentation and reddit threads and fairly certain the solution has been staring me in the face, but I'm not seeing how to do it. 

I only have two layers on my Keebio Viterbi Rev2, QWERTY and RAISE. Currently, RAISE is only activated while holding the RAISE key. I want the layer to persist after releasing RAISE, and then return to QWERTY when I press RAISE again. Seems so simple, but I can't see how to do it.

What needs to be updated in the code of my keymap? 
[image]

Thank you for any help!

noroadsleft, 2020-01-13:
1. Delete the entire `process_record_user()` function. You don't need it. (Lines 25-42)
2. Delete the `custom_keycodes` enum. Don't need that either. (Lines 4-7)
3. Insert `#include QMK_KEYBOARD_H` as the first line of the file.
4. Change both instances of the `RAISE` keycode to `TG(_RAISE)`. (Currently Lines 13 and 20)  
   `TG()` toggles the state of the given layer – if it's off, the layer turns on, and if it's on, the layer turns off.

[Advanced Keycodes: Switching and Toggling Layers - QMK Firmware Docs](https://docs.qmk.fm/#/feature_advanced_keycodes?id=switching-and-toggling-layers)

pmc222, 2020-01-13:
@noroadsleft - Thank you so much for your help on my last two issues! Worked like a charm.

closed 2020-01-13 by pmc222