qmk/qmk_firmware #7884: [Feature Request] Add change detection to qmk cformat

skullydazed, 2020-01-13:
Currently `qmk cformat` without any additional arguments runs clang-format against everything. To make it easier and faster to use we should support detecting changes and run only against those files which have been edited.

## Feature Request Type

- [X] Core functionality
- [X] Alteration (enhancement/optimization) of existing feature(s)

## Description

When the user supplies a list of filenames (by running `qmk cformat <file1> <file2> <...> <fileN>`) the behavior should be unchanged.

When the user runs `qmk cformat -a` the command should behave exactly as running `qmk cformat` does today, by running `clang-format` against all core files.

When the user runs `qmk cformat` without any arguments we should use the output of `git diff --name-only origin/master <core directories>` to find the files to format. We should provide a new optional option `--master-branch` (`-b`) to specify a branch other than `origin/master`.

skullydazed, 2020-01-13:
The behavior is still a WIP. I've written these down as a starting point but if there's an argument for doing it differently please suggest changes. For example, I'm not entirely fond of the name `--master-branch` but I'm wary of the implications of having `user.branch` overriding behavior here.

noroadsleft, 2020-01-13:
I'd suggest `--base-branch` instead of `--master-branch`, because `base branch` is GitHub's term for the target of a pull request. The default branch of a repo isn't always going to be named `master`. `-b` alias still works IMO though.

skullydazed, 2020-01-13:
I like that name a lot better.

PR #7893 (2020-02-23): New flags and functionality for cformat

## Description

`qmk cformat <file1> <file2> <...> <fileN>` functionality remains unchanged
`qmk cformat -a` formats all core files, same as old `qmk cformat` behavior
`qmk cformat` now only formats files that have changed between local and default `origin/master`. Branch can be changed via the `-b` flag.

## Types of Changes

- [X] Core
- [X] Enhancement/optimization

## Issues Fixed or Closed by This PR

* Fixes #7884 

## Dependencies

* #7872 

## Checklist

- [X] My code follows the code style of this project.
- [X] My change requires a change to t [... cut]
closed 2020-02-23 by skullydazed