qmk/qmk_firmware #7899: NK65 Indicator LEDs flash briefly but do not stay lit

dukrous, 2020-01-14:
Since the refactor (https://github.com/qmk/qmk_firmware/pull/7862) was done, NK65 Indicator LEDs now briefly flash and then stay dark whereas before they would stay lit.

If I revert the NK65 code to before the refactor the indicators stay lit, so this appears to be the direct and only cause.

zvecr, 2020-01-14:
@yiancar can you take a look into this please.

yiancar, 2020-01-14:
Hello,
Could you please update after https://github.com/qmk/qmk_firmware/pull/7895 ?
Also before you flash, please use the escape method (bootmagic light) to get in the bootloader.

If you have flashed a .bin with before https://github.com/qmk/qmk_firmware/pull/7862 then:
Unplug USB, press hold esc, plug usb, wait 5 seconds, release esc flash:)

dukrous, 2020-01-14:
Hey,

I did have a .bin before #7862 so I followed the steps. Wiping with a new build produced the same issue. If I revert back to the known good .bin then everything is happy.

yiancar, 2020-01-14:
Hm thats very weird if you can add me on discord yiancar#1180 so we can do some more debug that would be great :)

dukrous, 2020-01-14:
Request sent.

yiancar, 2020-01-14:
Note that this is only happening in LED_OFF, LED_SINGLE_COLOR, LED_DOUBLE_COLOR

closed 2020-01-14 by fauxpark