qmk/qmk_firmware #7913: [Bug] Encoder rapid-firing on right (slave) half of Iris Rev. 4

WindowDump, 2020-01-16:
When compiling new firmware for my Keebio Iris Rev. 4 based on QMK 0.7.109, I have found the left (master) half's encoder is reversed, while the right (slave) half's encoder fires both the CW and CCW mapped keys very rapidly over a period of a few seconds.

## Describe the Bug

Using encoder handling code based on the default, when the right encoder is turned, instead of sending the desired key code, it will fire both, with some random variation at the start before sending a large amount of the undesired key code, then a large amount of the desired keycode.

Example: if 3 is sent when a CW rotation is reported, and 4 is sent when a CCW rotation is reported, on a CCW rotation the keyboard sends:

``44333333333333333333333333333333333333333333333333333333333333333333333333333333333333333344444444444444444444444444444444444444444444444444444444444444444444444444444444444444``

If ``git revert 201c5bfa5cdce723f148b6e9a276219c8f2e8613`` is run before making firmware, the encoders behave as expected. (Thanks to [drashna's post on /r/olkb](https://www.reddit.com/r/olkb/comments/eiogxt/help_rev_4_iris_rightside_slave_encoder/fcspkcj/)!)

## System Information

 - Keyboard: Keebio Iris
   - Revision (if applicable): Rev. 4
 - Operating system: Windows 10 Education x64
 - AVR GCC version: 8.3.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.7.109

 - Any keyboard related software installed? 
   - [x] AutoHotKey (not currently running)
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I've been careful to clean old build artifacts before running building and flashing the keyboard, with ``make clean keebio/iris/rev4:default:flash`` (or similar) used instead of simply making.

drashna, 2020-01-16:
@nooges and @XScorpion2, tagging you due to related code.

WindowDump, 2020-01-16:
I get the result when I rotate slowly - just one dent CCW, then pausing. The behavior when turning it faster is a lot more chaotic. It eventually stops on its own after about 3 or 4 seconds.

1. I get 88 instances of 4 down, I think. When I focus a text editor and turn the encoder, there's 2 at the start (before the 3's), and 86 afterwards. (The string I posted earlier showed up in two separate instances.)
2. It eventually stops after I let got of the encoder.
3. It's alternating 3 and 4.
4. There's a few 4's, then a large batch of 3's, then another batch of 4's.
5. I'm using the encoder sold through the Keebio store, [an EC-11](https://keeb.io/products/rotary-encoder-ec11). Same on both halves.

I can attempt to flash the firmware with the issue on my right half now that I have a good firmware on my left one, though that might cause the issue to happen on the left half instead.

XScorpion2, 2020-01-16:
Wait, did you only flash your main hand?
The encoder change was a change in the way the halves communicate encoder state, so you need to flash both halves.

WindowDump, 2020-01-16:
Ah, I didn't see any changes to the docs to do that, and I haven't had to do that in the past.

Just flashed both halves, and I don't get the random firing anymore, though now the encoders are both flipped. I can use the undef encoders hack in the meantime until that other fix goes in, and I'll be sure to flash both halves next. Thanks!

closed 2020-01-16 by WindowDump