qmk/qmk_firmware #7916: [Bug] EECONFIG_RGB_MATRIX_SPEED is unused

robhaswell, 2020-01-16:
`EECONFIG_RGB_MATRIX_SPEED` is defined but never read. This is less of a bug report and more of a manual linting error.

## Describe the Bug

`EECONFIG_RGB_MATRIX_SPEED` appears in the following lines:
* https://github.com/qmk/qmk_firmware/blob/320822d75b785401809f45007320e6fb6885b3fd/tmk_core/common/eeconfig.h#L48
* https://github.com/qmk/qmk_firmware/blob/a91c0c476507cb8c12840abb59bff34ab0de3c03/tmk_core/common/eeconfig.c#L49
* Additionally it affects the value here: https://github.com/qmk/qmk_firmware/blob/320822d75b785401809f45007320e6fb6885b3fd/tmk_core/common/eeconfig.h#L52

It appears that this value is never read except to store to the eeprom - it is never read out of the eeprom. Additionally the animation speed is now a property of the `rgb_matrix_config` struct. I believe this is a leftover from the RGB matrix overhaul and should be removed.

robhaswell, 2020-01-16:
There is also a comment `// TODO: Combine these into a single word and single block of EEPROM` which seems redundant if these references are removed.

fauxpark, 2020-01-17:
This has to do with the size of the RGB Matrix config struct. From rgb_matrix_types.h:
```c
typedef union {
    uint32_t raw;
    struct PACKED {
        uint8_t enable : 2;
        uint8_t mode : 6;
        HSV     hsv;
        uint8_t speed;  // EECONFIG needs to be increased to support this
    };
} rgb_config_t;
```
`enable` and `mode` make up a single byte (the numbers after them denote the amount of bits allocated). `hsv` is three bytes, and of course `speed` is a byte. That's five bytes all up.

But wait... this is supposed to fit into the four bytes of a uint32! The RGB Matrix code gets around this by using `eeprom_read/update_block()` and passing in the size of the struct, instead of `eeprom_read/update_dword()`. Thus `EECONFIG_RGB_MATRIX_SPEED` is defined at EEPROM address 32, even though there are no functions that actually operate on just that specific byte.

As for the comment below it, that is a similar situation which relates to this PR: https://github.com/qmk/qmk_firmware/pull/6110
wherein some new entries were added to the `keymap_config` struct, however all the bits of the raw byte were already allocated, so it had to be split up into two non-consecutive bytes in EEPROM.

robhaswell, 2020-01-17:
Ah I see, thanks for explaining.

closed 2020-01-17 by robhaswell