qmk/qmk_firmware #7920: QMK Cli minimum version

Curry, 2020-01-16:
Proposed by @Erovia : Updating our minimum python version from 3.5 to 3.6 would allow us to take advantage of:
* F-Strings: [PEP 498](https://www.python.org/dev/peps/pep-0498/)
* PathLib: [PEP 428](https://www.python.org/dev/peps/pep-0428/), Specifically in 3.6, will allow us to walk directories and retrieve all sub-folders and their files using pathlib [PEP 519](https://www.python.org/dev/peps/pep-0519/)

Others?

To carry this out, we need to know the minimum supported versions of python 3 on all our supported platforms, going back to the oldest LTS version. Here is a table of our supported distributions, and the minimum available python version:

| Platform | Python Version |
| ------------- |-------------|
| Windows | 3.8.1 (MSYS) |
| MacOS | 3.7.6 (Homebrew) |
| FreeBSD | ? |
| WSL | ? |
| Arch | 3.8.1 (Arch Packages) |
| Manjaro | 3.8.1 (Arch Packages) |
| Fedora | ? (31), ? (30) |
| Ubuntu | 3.5.1 (16.04 LTS), 3.6.5 (18.04 LTS) |
| Debian | 3.7.3 (10), 3.5.3 (9), 3.4.2 (8) |
| Gentoo | 3.6.9 (Note: Install script pins to 3.5, version no longer available) | 
| Sabayon | ? (dev-lang/python, no install version specified) |
| openSUSE Tumbleweed | 3.7.3 (Official) |
| openSUSE Leap 15.1 | 3.6.9 (Official) |
| Slackware | 3.8.1 (Slackware current) |
| Solus | ? |
| Void | 3.8.1 |

This is the ones I could publicly find, if anyone is running one of these distributions that have a question mark, please comment it below.

As of right now, Ubuntu 16.04 LTS Xenial, Debian 8, and Debian 9 has <3.6 python support. 
Additionally, the qmkfm/base_container python version is using Debian 9.

Erovia, 2020-01-16:
Both Fedora 30 and 31 has 3.7+.

Erovia, 2020-01-16:
It seems FreeBSD also has [3.7+](https://www.freebsd.org/cgi/ports.cgi?query=^python37-3.7.6&stype=name).

Erovia, 2020-01-16:
Solus is currently on [3.6.9](https://dev.getsol.us/source/python3/) with plans to 3.7+.

Erovia, 2020-01-16:
Sabayon: [3.6.9](https://packages.sabayon.org/show/python,240084,sabayon-weekly,amd64,5,standard)

Erovia, 2020-01-16:
For WSL I believe it depends on the installed userspace, but I'm not a Win user.

zvecr, 2020-05-10:
Completed in #8835

closed 2020-05-10 by zvecr