qmk/qmk_firmware #7931: [Bug] Skog Lite default keymap is incorrect

kyleduvall, 2020-01-18:
The default keymap for the Percent Skog Lite is incorrect and has several keys flipped around.

## Describe the Bug
Below are all the keys that aren't operating correctly.

Expected | Actual
-- | --
F11 | F12
F12 | F11
\| | Delete
Right Shift | Up
Up | Right Shift
Right Ctrl | Down
Down | Right Ctrl
Ins | Backspace
Home | Pause
Delete | \|
Pause | Home

## System Information

 - Keyboard:  Percent Skog Lite
   - Revision (if applicable):  
 - Operating system: Windows and OSX
 - AVR GCC version: 8.3.0

 - ARM GCC version: 7.3.1

 - QMK Firmware version: 0.7.109

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

mechmerlin, 2020-01-18:
@kyleduvall flashed a WKL ANSI board with the default keymap which is standard TKL. I've asked him to reflash with a keymap suitable for his layout and see if the keymap is still wrong. I suspect this might be an issue with the physical matrix as when this board was ported over, I had a broken PCB.

kyleduvall, 2020-01-18:
> @kyleduvall flashed a WKL ANSI board with the default keymap which is standard TKL. I've asked him to reflash with a keymap suitable for his layout and see if the keymap is still wrong. I suspect this might be an issue with the physical matrix as when this board was ported over, I had a broken PCB.

@mechmerlin per your recommendation I tried compiling and flashing using `LAYOUT_tkl_7u_ansi_blocker` this produced the same result. I also tried `LAYOUT_tkl_7u_ansi` and that had the same problem as well. At this point I've tried all supported layouts and they all produce the same issue. I'm thinking you may be right about it being an issue with the physical board you were given to port it.

brettinman, 2020-01-19:
I think I fixed this by simple swapping two columns. Works on my board with the default layout but would be great if someone else with a board could test.

https://github.com/qmk/qmk_firmware/pull/7940

kyleduvall, 2020-01-19:
> I think I fixed this by simply swapping two columns. Works on my board with the default layout but would be great if someone else with a board could test.
> 
> #7940

I can confirm that @brettinman's fix resolves the issues on my Skog Lite.

closed 2020-01-22 by mechmerlin