qmk/qmk_firmware #7954: [Bug]  `make` doesn't follow symlinks

C0DK, 2020-01-21:
## Describe the Bug
I am trying to keep version control of my own firmware in my dotfiles, and the way i wanted to do that was to symlink the dotfile-folder into the firmware folder, so `qmk_firmware/keyboards/preonic/keymaps/c0dk` points to `dotfiles/qmk_keyboard`.. 

However `make preonic/rev3:c0dk` returns `make[1]: *** No rule to make target 'cabang'.  Stop.` - which is naturally because it doesn't follow symlinks... 
Not sure if this would be a new feature or a bug..

But it could help fix https://github.com/qmk/qmk_firmware/issues/1120 - because i can then keep the firmware repo with non-checked-out files, and simply use a VCS on my own folder.. but I need `make`to follow symlinks.. which it doesn't.

## System Information

 - Operating system: Manjaro Linux x86_64
 - AVR GCC version: 9.2.0
 - ARM GCC version: 9.2.0
 - QMK Firmware version: 0.7.105

ridingqwerty, 2020-01-24:
This worked fine for me:

```
cp -r keyboards/atreus/keymaps/ridingqwerty ~/ridingqwerty
rm -rf ridingqwerty
ln -s ~/ridingqwerty keyboards/atreus/keymaps/ridingqwerty
make atreus:ridingqwerty
```

What's jumping out to me here is that the keymap name is **cd0k**, but `make` chokes on rule to make **cabang** -- what is that a reference to?

C0DK, 2020-01-25:
oops sorry it's because I call it "cabang" but wanted to change it to make it obvious that i was my username. I'll try again i guess.

ridingqwerty, 2020-01-26:
I think as long as the symlink is sound, e.g. the target and destination names match, represent the named `<keymap>` directory, and relevant keymap structure is there, it should be fine.

C0DK, 2020-01-28:
You were right. i think i placed the symlink the wrong place. it works now! thanks!

closed 2020-01-28 by C0DK

ridingqwerty, 2020-01-28:
Awesome!

andresilva, 2020-10-21:
This is not working for me. Could any of you confirm if this still works or if something has changed in the build process in the meantime?

ridingqwerty, 2020-10-21:
The procedure I used [here](https://github.com/qmk/qmk_firmware/issues/7954#issuecomment-578155654) still works for me (I amended it slightly to make the `rm` more explicit after copying the keymap out of tree).

What exactly are you doing?

andresilva, 2020-10-21:
This is what happens to me if I follow your instructions:

```
> cd /tmp

> git clone git@github.com:qmk/qmk_firmware.git
Cloning into 'qmk_firmware'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 221551 (delta 1), reused 1 (delta 1), pack-reused 221547
Receiving objects: 100% (221551/221551), 148.22 MiB | 14.87 MiB/s, done.
Resolving deltas: 100% (139623/139623), done.

> cd qmk_firmware

> cp -r keyboards/atreus/keymaps/ridingqwerty ~/ridingqwerty; \
rm -rf keyboards/atreus/keymaps/ridingqwerty; \
ln -s ~/ridingqwerty keyboards/atreus/keymaps/ridingqwerty

> ll keyboards/atreus/keymaps/ridingqwerty
lrwxrwxrwx 1 andre users 24 Oct 21 16:28 keyboards/atreus/keymaps/ridingqwerty -> /home/andre/ridingqwerty

> make git-submodule
...

> make atreus:ridingqwerty
QMK Firmware 0.10.41
WARNING:
 Can not run bin/qmk! This tool will be required when the develop branch is merged on 2020 Aug 29.

 Please run util/qmk_install.sh to install all the dependencies QMK requires.

Making atreus/astar with keymap clash and target ridingqwerty                                  [ERRORS]
make[1]: *** No rule to make target 'ridingqwerty'.  Stop.
Making atreus/astar with keymap classic and target ridingqwerty                                [ERRORS]
make[1]: *** No rule to make target 'ridingqwerty'.  Stop.
Making atreus/astar with keymap default and target ridingqwerty                                [ERRORS]
make[1]: *** No rule to make target 'ridingqwerty'.  Stop.
...
```

andresilva, 2020-10-21:
The only way it works for me is if I copy the folder and instead symlink all the files inside it (which is a bit less pratical than just symlinking the folder).

ridingqwerty, 2020-10-21:
Can you make anything else?  What does this do?

`make planck/rev6:default`

andresilva, 2020-10-21:
```
❯ make planck/rev6:default
QMK Firmware 0.10.41
WARNING:
 Can not run bin/qmk! This tool will be required when the develop branch is merged on 2020 Aug 29.

 Please run util/qmk_install.sh to install all the dependencies QMK requires.

Making planck/rev6 with keymap default

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/planck/planck.c                                                                [OK]
Compiling: keyboards/planck/rev6/rev6.c                                                             [OK]
...
```

ridingqwerty, 2020-10-21:
I wonder if this is something to do with the group ownership of `/tmp` being `root`.  In my example everything in `qmk_firmware` and `~` is owned by my user.

andresilva, 2020-10-21:
My initial tests were done in my home, I only cloned to `/tmp` to show the full setup. Are you using Linux?

ridingqwerty, 2020-10-21:
Yeah I'm on Ubuntu 18.04

andresilva, 2020-10-21:
I believe it was caused by this bug https://savannah.gnu.org/bugs/?func=detailitem&item_id=53465. I was using GNU Make 4.2.1, with 4.3 it works.

babramczyk, 2021-05-13:
Having the same issue, on macOS

I’ve symlinked using the following command (from my dotfiles repository):

```terminal
ln -s qmk/keymaps/babramczyk ../qmk_firmware/keyboards/preonic/keymaps/babramczyk
```

And I’ve made sure that `<dotfiles_repo_path/qmk/keymaps/babramczyk` has all of original files that were in the `babramczyk` folder in the `qmk_firmware` repo (i.e. `keymap.c`).

I’ve also download the most up-to-date `make` from GNU (`4.3`) (which, on macOS, by installing through homebrew, is called `gmake`).

This is what happens when I try to compile (first one specifies no `rev3`, second one does):

```terminal
➜  qmk_firmware git:(master) ✗
  gmake preonic:babramczyk
QMK Firmware 0.12.48
make: *** No rule to make target 'preonic:babramczyk'. Stop.
|
|  QMK's make format recently changed to use folder locations and colons:
|     make project_folder:keymap[:target]
|  Examples:
|     make dz60:default
|     make planck/rev6:default:flash
|

➜  qmk_firmware git:(master) ✗
  gmake preonic/rev3:babramczyk
QMK Firmware 0.12.48
Making preonic/rev3 with keymap 0xdec and target babramczyk                                    [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap 333fred and target babramczyk                                  [ERRORS]
Enabling LTO on ChibiOS-targeting boards is known to have a high likelihood of failure.
If unsure, set LTO_ENABLE = no.
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap AlexDaigre and target babramczyk                               [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap CMD-Preonic and target babramczyk                              [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap alfrdmalr and target babramczyk                                [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap arkag and target babramczyk                                    [ERRORS]
Enabling LTO on ChibiOS-targeting boards is known to have a high likelihood of failure.
If unsure, set LTO_ENABLE = no.
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap badger and target babramczyk                                   [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
Making preonic/rev3 with keymap bghull and target babramczyk                                   [ERRORS]
gmake[1]: *** No rule to make target 'babramczyk'.  Stop.
```

The second one is truncated for briefness, but it just seems to try to build a keymap for every single keymap in `keyboards/preonic/keymaps`, instead of building just the keymap for `babramczyk`.

I’ve also confirmed that removing the symlink, and putting all the of files physically back into the `qmk_firmware` repo works as intended. With both `make` (BSD make that comes with macOS, v3.81) and `gmake` (GNU make from homebrew, v4.3).

Any thoughts or advice? 😀

fauxpark, 2021-05-14:
QMK doesn't really support symlinks, since as you can see it doesn't really work right all the time, and is especially problematic on Windows.