qmk/qmk_firmware #7961: Nyquist Rev 3 Slave Underglow Issues

pmc222, 2020-01-22:
I can't get the slave half of my Nyquist Rev 3 to change from anything but a red underglow. I've read through the documentation and a bunch of reddit threads, but just can't figure it out . I have flashed both master and slave side independently of each other, as well as while connected through the TRRS cable. In order, keymap.c, congig.h, and rules.mk files from my custom keymap folder screenshots below:

[image]

All changes will only reflect on the side that is plugged in. I've changed the underglow color (green) and also tried updating the RGBLIGHT_VAL_STEP value, which work on the plugged in half. The side that is not plugged always has an underglow color of red and a RGBLIGHT_VAL_STEP of 8. 

After getting frustrated with not being able to figure it out, I went ahead and disabled RGBLIGHT_ENABLE and even that would only work on the plugged in half, so I would have half of my keyboard red and half of it with no underglow. 

What do I need to do to get both sides to work together?

Typing on both halves works without issue.

zvecr, 2020-01-22:
IIRC, you need to use the `rgblight_sethsv` set of functions, not the _rgb ones. Its also recommended to use the no eeprom versions, `rgblight_sethsv_noeeprom`, to avoid wear to the MCUs memory.

pmc222, 2020-01-22:
Again, thank you. This was driving me crazy. Lifesaver!

pmc222, 2020-01-22:
@zvecr - question: the following updated the entire keyboard as mentioned above:
void led_set_user(uint8_t usb_led) {
  rgblight_sethsv_noeeprom(HSV_GREEN);
}

but when trying to set underglow by layer with the following, it would only change the master half:
uint32_t layer_state_set_user(uint32_t state) {
  switch(biton32(state)) {
    case _QWERTY:
      rgblight_sethsv_noeeprom(HSV_GREEN);
      break;
    case _RAISE:
      rgblight_sethsv_noeeprom(HSV_ORANGE);
      break;
  }
  return state;
}

Any ideas? Thank you again for your help!

closed 2020-05-21 by stale[bot]