qmk/qmk_firmware #7962: How to get TT(Layer) to work - Nyquist Rev 3

pmc222, 2020-01-22:
The toggle function of TT(layer) is not working for me. I can only get it to respond like MO(layer). I've tried adjusting #define TAPPING_TOGGLE to 2 and then 1, with no change. Keymap.c screenshot below. How do I get this to work?

[image]

zvecr, 2020-01-22:
`#define TAPPING_TOGGLE 2` needs to go in a `config.h`.

Something like `keyboards/keebio/nyquist/keymaps/pmc222/config.h`, which contains:
```c
#pragma once 

#define TAPPING_TOGGLE 2
```

pmc222, 2020-01-22:
Oh wow, silly mistake. Thank you so much! That fixed the issue.

closed 2020-01-22 by pmc222