qmk/qmk_firmware #7964: i cannot build ergodox_infinity:default (chibios?)

vikke, 2020-01-22:
i tried to clone qmk_firmware and `make git-submodule` and `make ergodox_infinity:default`.

so below errors occured.

```
Compiling: lib/chibios/os/common/ports/ARMCMx/chcore.c                                              [OK]
Compiling: lib/chibios/os/common/ports/ARMCMx/chcore_v7m.c                                         In file included from ./lib/chibios-contrib/os/common/ext/CMSIS/KINETIS/k20x7.h:134,
                 from ./lib/chibios-contrib/os/common/startup/ARMCMx/devices/K20x7/cmparams.h:70,
                 from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70,
                 from ./lib/chibios/os/rt/include/ch.h:81,
                 from lib/chibios/os/common/ports/ARMCMx/chcore_v7m.c:28:
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function 'SVC_Handler':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function '_port_irq_epilogue':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_ergodox_infinity/lib/chibios/os/common/ports/ARMCMx/chcore_v7m.o] Error 1
make: *** [ergodox_infinity:default] Error 1
Make finished with errors
```

how to do it?

Curry, 2020-01-22:
You need to downgrade your `arm-none-eabi-gcc` to 8.3

vikke, 2020-01-22:
@Curry thanks. 
I'll try to download it at https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads.

Curry, 2020-01-22:
What is your operating system? There's probably a way to pin to a version using your package manager

vikke, 2020-01-22:
i'm using MacOS. and using Home Brew.
i can't find old version's(pinning) package.

Curry, 2020-01-22:
You should uninstall arm-none-eabi-gcc, and reinstall with 
`brew install arm-none-eabi-gcc@8`

fauxpark, 2020-01-22:
Make sure your QMK repo is up to date, uninstall your existing arm toolchain, then run `./util/qmk_install.sh`. It will install the correct version.

vikke, 2020-01-22:
@fauxpark my problem was resolved. thank you.

closed 2020-01-22 by vikke