qmk/qmk_firmware #7988: Can combos be declared on user files?

danielo515, 2020-01-23:
Hello.
I wanted to start using combos. Because the documentation is not clear about what `config.h` file should I modify I just took for granted that I can define combos on my user configurations.

However, doing that results in compilation errors. Then I moved combo definitions to my particular keymap and then everything worked as expected.
Does this mean that combos are impossible to define on a global way on a user file? I want my combos to be shared across all my keyboards.

Regards

ridingqwerty, 2020-01-24:
Hey @danielo515, just want to confirm whether you figured this out?  I know we talked a bit on Discord earlier today.

danielo515, 2020-01-24:
I didn't tried again.
But maybe you can tell me here the proper files to modify?
What I did was:

- added my combos to user.c file
- added required declarations (definnes) to user.h
- added COMBO_ENABLE = yes to the keyboard where I want combos to be enabled
- Tried to compile that keyboard and see errors about combo stuff not being declared

Maybe I did something wrong at any of those steps?

danielo515, 2020-01-24:
By the way, why can I not add the define of combo count to the same file where I have the combo definitions? It is a bit painful to have them split like this.

danielo515, 2020-01-24:
This are the errors that I get if I try to definne the combos on my user files:

```
s/danielo515/danielo515.c:1:0:
users/danielo515/danielo515.h:187:0: error: "COMBO_COUNT" redefined [-Werror]
 #define COMBO_COUNT 4
 ^
In file included from quantum/quantum.h:119:0,
                 from users/danielo515/danielo515.h:4,
                 from users/danielo515/danielo515.c:1:
quantum/process_keycode/process_combo.h:51:0: note: this is the location of the previous definition
 #define COMBO_COUNT 0
 ^
In file included from users/danielo515/danielo515.c:1:0:
users/danielo515/danielo515.h:188:0: error: "COMBO_TERM" redefined [-Werror]
 #define COMBO_TERM 150
 ^
In file included from quantum/quantum.h:119:0,
                 from users/danielo515/danielo515.h:4,
                 from users/danielo515/danielo515.c:1:
quantum/process_keycode/process_combo.h:54:0: note: this is the location of the previous definition
 #define COMBO_TERM TAPPING_TERM
 ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_ergodox_ez_danielo515/danielo515.o] Error 1
tmk_core/rules.mk:377: recipe for target '.build/obj_ergodox_ez_danielo515/danielo515.o' failed
Makefile:542: recipe for target 'ergodox_ez:danielo515' failed
Make finished with errors
make: *** [ergodox_ez:danielo515] Error 1
```

danielo515, 2020-01-24:
Ok, after a close analysis to what the user kuchosauronad0 did, I came to the conclusion that the definitions needs to be added to config.h on my user directory. I was adding them just to danielo515.h
Now it is working. Thanks

closed 2020-01-24 by danielo515

ridingqwerty, 2020-01-24:
I figured it was something like that.  Glad you figured it out.