qmk/qmk_firmware #8003: [Feature Request] allow combos with dual function keys

danielo515, 2020-01-25:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I just discovered combos and I'm amazed by the number of posibilities they brinng to my home row. However, most of my home row keys are dual function keys (h for meh, j for alt, k for control...) so I can't use them for combos. Is it possible to allow such kind of keys on combos? and if it not possible for now, will it be possible to implement?

drashna, 2020-02-08:
I think you need to define `COMBO_ALLOW_ACTION_KEYS`, and then use the whole keycode.

danielo515, 2020-02-09:
Wow, didn't noticed that in the docs. Of course I suppose it is documented, but I didn't found it.
Thanks, I'll try it out for sure

danielo515, 2020-02-16:
Thanks for pointing it out @drashna . I looked for it on the docs and there is just one line that doesn't even make clear what it is that configuration for:
> You may also be able to enable action keys by defining COMBO_ALLOW_ACTION_KEYS.

I tried it and, indeed it works, but it has severe drawbacks, which is that the keys you use for a combo no longer work for as dual function keys (in my case).

closed 2020-06-15 by stale[bot]